foxglove/studio #5314: Bad error messages on incorrectly encoded flatbuffer messages

jkuszmaul, 2023-02-09:
**Description**

I was converting a websocket server to flatbuffers and accidentally started base64 encoding my message data unnecessarily. When I opened the raw message panel, I'd see the timestamp update on every new message but there were no errors indicating why I couldn't see any data (whether in the console or in the panel).

**Steps To Reproduce**
Let me know if I can put together code for this usefully. But:

* Setup a websocket server.
* Use a valid flatbuffer schema.
* Send an invalid flatbuffer message (can be just zeros or something).
* Have a Raw Message panel open.
* Observe timestamps updating, but no data contents and no errors.

**Expected Behavior**
An error, at least in the developer console, to enable faster debugging.

snosenzo, 2023-02-15:
@jkuszmaul 
so after looking into this it looks like this might be because of the `Parser` class's `toObject` method. The result of this function ends up being an empty object because all of the `read` functions return null and just fail to be added to the object instead of erroring out.   https://github.com/jkuszmaul/flatbuffers_reflection/blob/main/src/reflection.ts#L211

so it returns an empty object which is then normalized by whatever panel consumes it into something consumable. 
So either I can check to see if all of the `fields` are populated on the output object within foxglove studio or I can open a PR against the `flatbuffers_reflection` repo to fail out of `toObject` when the result doesn't contain the read fields. I think it would make more sense for the `toObject` to fail out and studio to surface it. 

A question I have about this approach is whether we need to support partial flatbuffer messages? If so we'll have to find another way to validate

jkuszmaul, 2023-02-15:
@snosenzo I think you are right that this check should happen in the reflection library. I may need to think on exactly how those checks should happen and which things should and shouldn't be errors. I can take on addressing this, since I think I'll be pretty opinionated on exactly what should/shouldn't error. Thanks for tracking down the source.

The relevant issues will end up being (mostly notes for myself):
* Ensure that errors are thrown if the flatbuffer attempts to access memory that is out-of-bounds (I suspect that right now it silently returns null).
* Ensure that a valid root table is present (should always be the case).
* Using a newer version of the schema should always succeed at reading an older version of the flatbuffer (assuming the schemas has evolved in a backwards-compatible way).
* Using an older version of the schema to read a newer version of a serialized message should probably result in an error, although this is less clear-cut (I should check what the regular flatbuffers code does).

snosenzo, 2023-02-15:
Sounds good, let me know if there's anything else I can help with

defunctzombie, 2023-03-13:
@jkuszmaul Any update on this? Is there something we can do in studio to be more defensive here and surface the error or does the underlying library require an update?

jkuszmaul, 2023-03-13:
@defunctzombie Various work caused me to forget about this entirely. I don't know exactly when I'll get to it. It won't be much work for me, but it's a bit of a context switch. I assume it isn't urgent.

Unfortunately, I'm not sure if studio could have meaningfully distinguished between this issue and a user sending a literally empty message. I think that this has to happen in the underlying library.

jkuszmaul, 2023-03-13:
If you want to clean this bug up as not being relevant to studio itself, I don't mind. Created https://github.com/jkuszmaul/flatbuffers_reflection/issues/46 to track this myself.

defunctzombie, 2023-03-13:
> literally empty message.

Is this something worthy of a warning by itself? An empty message sounds like something failed does it not?

jkuszmaul, 2023-03-13:
Sorry, "empty" can mean multiple things. If the returned object from the parsing code is `null` or `undefined`, then yes I'd expect an error of some sort. But if the parser returns `{}` that could be a valid message. There's no reason the user couldn't send a message that just so happens to not have any fields populated--especially if a given user is leaning heavily on protobuf/flatbuffer defaults in their messages to reduced serialized message size.

defunctzombie, 2023-04-03:
@jkuszmaul Do you have a file we can use to trigger these conditions? Or some tips on how to generate one that would have the failure modes you expect the reflection library to handle?

jkuszmaul, 2023-04-03:
I don't have a specific file; just to discuss the specific conditions I listed above, in case anyone wants to try to look at this before me:

> * Ensure that errors are thrown if the flatbuffer attempts to access memory that is out-of-bounds (I suspect that right now it silently returns null).

This could be triggered by attempting to parse entirely random data.  To distinguish this from the next bullet, one could also test the first bullet by constructing a flatbuffer, but when you attempt to add anything that is a pointer to memory (e.g., the `addName()` call at https://github.com/jkuszmaul/flatbuffers_reflection/blob/main/src/reflection.test.ts#L48 ), populate a random value instead.

> * Ensure that a valid root table is present (should always be the case).

This could be triggered by attempting to parse entirely random data.

> * Using a newer version of the schema should always succeed at reading an older version of the flatbuffer (assuming the schemas has evolved in a backwards-compatible way).

Serialize a flatbuffer. Then add a field to the schema. Using the same serialized data but a reserialized .bfbs for the reflection data, it should parse successfully (the new field should just return empty or the default value, depending on the requested behavior).

> * Using an older version of the schema to read a newer version of a serialized message should probably result in an error, although this is less clear-cut (I should check what the regular flatbuffers code does).

Do the reverse of the above. It should successfully parse (this is consistent with existing flatbuffer behavior; I could make an argument for exposing some sort of error, however). Any new fields that were removed from the schema should not be present.

PR #5788 (2023-04-18): Update flatbuffers_reflection to 0.0.4
**User-Facing Changes**

Better error messages when an incorrectly serialized flatbuffer message is encountered.

**Description**

Resolves #5314 by raising an exception when an invalid flatbuffer is encountered, rather than simply returning an empty object.
commit 2a2c7210 (2023-04-18): Update flatbuffers_reflection to 0.0.4 (#5788)

**User-Facing Changes**

Better error messages when an incorrectly serialized flatbuffer message
is encountered.

**Description**

Resolves #5314 by raising an exception when an invalid flatbuffer is
encountered, rather than simply returning an empty object.
closed 2023-04-18 by jtbandes