foxglove/studio #5380: If there is a part of empty data in the bag file, jumping when playing to empty data will make the page stuck

Woodii1998, 2023-02-17:
**Description**
If there is a part of empty data in the bag file, jumping when playing to empty data will make the page stuck

- Version: foxglove online version
- Platform: web

**Steps To Reproduce**

https://user-images.githubusercontent.com/33593572/219618367-783cd30e-a736-4b2e-b2cd-a7fd677cef54.mp4

**Expected Behavior**
The bag will play normally after the jump

Woodii1998, 2023-02-17:
If you need this bag file:
https://drive.google.com/file/d/1jLV-jY9KeOdor6BbJSkBEwmX-AIluBdq/view?usp=sharing

jtbandes, 2023-02-23:
I'm not able to reproduce this issue. I tried uploading the bag file to foxglove data platform, and this is the behavior I see.
Can you provide more details or clearer reproduction steps?

https://user-images.githubusercontent.com/14237/220793714-b49c7d2c-c07e-47ab-805e-6b8f0c8247fd.mov

Woodii1998, 2023-02-23:
The fact that I can still reproduce it stably, I think the steps to reproduce it have the following points.
- Make sure the page has no data cache
- Play from the beginning
- Make sure you play to the blank data section before skipping (I play to about ```2023-02-06T02%3A47%3A51.769733341Z``` if you're not sure where you need to play to)

https://user-images.githubusercontent.com/33593572/220836138-0e5fb30e-4313-426f-a6fa-b3d05d941553.mp4

Woodii1998, 2023-02-23:
https://github.com/foxglove/studio/blob/main/packages/studio-base/src/players/IterablePlayer/IterablePlayer.ts#L671

I tried to troubleshoot the issue and then found some clues, but probably didn't catch the essence of the problem

Normally when jumping this._abort.signal=true
When the problem occurs this._abort.signal=false

I'm not sure if this is the cause of the problem but I hope it helps you guys

defunctzombie, 2023-03-01:
I've been able to reproduce some of the undesirable behavior and am taking a look. Thank you for sharing the test bag and more detailed analysis.

closed 2023-03-02 by defunctzombie

defunctzombie, 2023-03-03:
@Woodii1998 Thanks for reporting this and providing the test data to reproduce. The fix for this is deployed to `studio.foxglove.dev`. Please give it a try and see if your issue is resolved.

commit 24c8fc3b (2023-03-02): Exit the producer loop when aborting during stamp wait (#5466)

For some data sources, large gaps in data produce `stamp` results in
their message iterators. These stamp results indicate where the
underlying data source has moved its cursor. The BufferedIterableSource
compares such stamp results to the current playhead location to
determine whether it needs to buffer more data from the data source. The
logic for this comparison sits in a loop until the playhead indicates it
is time to read more data. This loop did not properly handle abort
signals which resulted in the "producer" thread of the [... cut]