foxglove/studio #5382: Cannot load local URDF file into 3D View

codegrafix, 2023-02-17:
**Description**

Hi, I am not able to load a URDF file @ 3D view from my local system. Drag and drop as well as adding the path to the url form is not working.
The only way I got it working is uploading the URDF file to a webserver and then paste the link.

- Version: Foxglove Studio v1.40.0
- Platform: Ubuntu 22.04

**Steps To Reproduce**

The file I am using right now is as follows:
`<robot name="scooter">

	

 	<link name="base_link">
		<visual>
		    <origin xyz="-0.3 0 -0.5" rpy="0 0 0"/>
			<geometry>
				<box size="1 0.26 0.085"/>
			</geometry>
			<material name="Cyan1">
	       		<color rgba="0 0.9 0.9 1.0"/>
	     	</material>
		</visual>
	</link>

 	<link name="front_wheel">
		<visual>
		    <origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
			<cylinder radius="0.1" length="0.07"/>
			</geometry>
			<material name="red2">
	       		<color rgba="1 0 0 1.0"/>
	     	</material>
		</visual>
	</link>

 	<link name="back_wheel">
		<visual>
		    <origin xyz="0 0 0" rpy="1.57079 0 0"/>
			<geometry>
			<cylinder radius="0.1" length="0.07"/>
			</geometry>
			<material name="red3">
	       		<color rgba="1 0 0 1.0"/>
	     	</material>
		</visual>
	</link>

 	<link name="mast">
		<visual>
		    <origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
			<cylinder radius="0.025" length="1"/>
			</geometry>
			<material name="Cyan1">
	       		<color rgba="0 0.9 0.9 1.0"/>
	     	</material>
		</visual>
	</link>

 	<link name="handle">
		<visual>
		    <origin xyz="0 0 0" rpy="0 0 0"/>
			<geometry>
			<cylinder radius="0.025" length="0.25"/>
			</geometry>
			<material name="red1">
	       		<color rgba="1 0 0 1.0"/>
	     	</material>
		</visual>
	</link>

	<joint name="mast_to_base" type="revolute">
	    	<parent link="base_link"/>
	    	<child link="mast"/>
	    	<axis xyz="0 0 1"/>
	    	<limit lower="-1" upper="1" effort="10" velocity="3"/>
	    	<origin xyz="0 0 0" rpy="0 -0.2 0"/>
	</joint>

	<joint name="mast_to_handle" type="fixed">
	    	<parent link="mast"/>
	    	<child link="handle"/>
		<origin xyz="0 0 0.5" rpy="1.57079 0 0"/>
	</joint>
		<joint name="mast_to_front_wheel" type="fixed">
	    	<parent link="mast"/>
	    	<child link="front_wheel"/>
		<origin xyz="0 0 -0.60" rpy="1.57079 0 0"/>
	</joint>

	<joint name="base_link_to_back_wheel" type="fixed">
	    	<parent link="base_link"/>
	    	<child link="back_wheel"/>
		<origin xyz="-0.8 0 -0.60" rpy="0 0 0"/>
	</joint>

</robot>`

I also tried it with other files, but the local method does not work at all.

**Expected Behavior**

url: It should be possible to directly pass the local file path to URDF Layer.

jtbandes, 2023-02-17:
It works for me in the desktop app using a `file:///` URL:

[image]

codegrafix, 2023-02-18:
Thank you. This works!

closed 2023-02-18 by codegrafix