foxglove/studio #5455: Custom panel - Cannot subscribe to certain topics and republish that same message

joshnewans, 2023-03-01:
**Description**

- Version: 1.42.0
- Platform: Ubuntu 20.04 (Pop OS) x64

For certain message types (tested only with ROS 2 `sensor_msgs/Joy`), an attempt to republish a message received from a subscription will fail silently. Put another way, it appears the serialise/deserialise (if you can call it that) is not quite compatible for sub/pub.

As I understand it, in this case the message contains an array of floats and an array of ints. Foxglove parses these from the subscriber as a `Float32Array` and `Int32Array` respectively, but the publisher will only accept raw arrays. A simple JSON wrap doesn't work, as it converts the special types into objects with numerical keys rather than actual arrays. Here is my current workaround.

```
const tmpMsg = {
    header: latestMsg.header,
    axes: Array.from(latestMsg.axes),
    buttons: Array.from(latestMsg.buttons),
};
```

**Steps To Reproduce**
An example repo is provided below with details and instructions in the README. Please note that this repo also demonstrates some other issues I am having with publisher timing (I suspect that one is user error) so if any assistance can be provided on that too that would be appreciated.
https://github.com/joshnewans/fox-pub-error-mwe

Essentially, subscribe to a `sensor_msgs/Joy` topic e.g. `/joy` (that is fed by external data), advertise a new topic of that same type (e.g. `/joy2`), and when receiving a message publish it directly.

**Expected Behavior**

One would expect to receive messages on the new topic, but they will not appear. Instead a conversion like the one described above is required.

jtbandes, 2023-03-02:
Thanks for the report and detailed writeup. I've made a PR https://github.com/foxglove/studio/pull/5473 which I believe addresses the other issues you reported in the repo readme, but not this one (about TypedArrays).

I also tweaked your usage of React to avoid side effects in setState updater functions. The changes I used for testing were:

```diff
diff --git a/src/ExamplePanel.tsx b/src/ExamplePanel.tsx
index 5785736..5bd1afc 100644
--- a/src/ExamplePanel.tsx
+++ b/src/ExamplePanel.tsx
@@ -32,8 +32,6 @@ function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Elem
   const [messages, setMessages] = useState<readonly MessageEvent<unknown>[] | undefined>();
   const [rawMsg, setRawMsg] = useState<Joy | undefined>();
   const [convMsg, setConvMsg] = useState<Joy | undefined>();
-  const [, setSubTopic] = useState<string | undefined>();
-  const [pubTopic, setPubTopic] = useState<string | undefined>();
 
   const [renderDone, setRenderDone] = useState<(() => void) | undefined>();
 
@@ -100,7 +98,6 @@ function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Elem
 
   // We use a layout effect to setup render handling for our panel. We also setup some topic subscriptions.
   useLayoutEffect(() => {
-
     context.onRender = (renderState: RenderState, done) => {
       setRenderDone(() => done);
       setTopics(renderState.topics);
@@ -113,18 +110,13 @@ function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Elem
     context.watch("currentFrame");
 
     // COMMENT OUT THE LINE BELOW
-    context.advertise?.("/joy2", "sensor_msgs/msg/Joy");
-
+    // context.advertise?.("/joy2", "sensor_msgs/msg/Joy");
   }, [context, config]);
 
-
   // Subscribe to the topic
   useEffect(() => {
     console.log("Subscribe to " + config.subTopic);
-    setSubTopic(() => {
-      context.subscribe([config.subTopic]);
-      return config.subTopic;
-    });
+    context.subscribe([config.subTopic]);
   }, [config.subTopic, context]);
 
   // Process messages
@@ -143,40 +135,27 @@ function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Elem
     }
   }, [messages]);
 
-
   // Advertise the publish topic
   useEffect(() => {
-    console.log("About to advertise publisher on " + config.pubTopic);
     if (config.publishOn) {
-      setPubTopic((oldTopic) => {
-        if (oldTopic) {
-          context.unadvertise?.(oldTopic);
-        }
-        context.advertise?.(config.pubTopic, "sensor_msgs/msg/Joy");
-        return config.pubTopic;
-      });
+      console.log("About to advertise publisher on " + config.pubTopic);
+      context.advertise?.(config.pubTopic, "sensor_msgs/msg/Joy");
+      return () => context.unadvertise?.(config.pubTopic);
     }
+    return undefined;
   }, [config.publishOn, config.pubTopic, context]);
 
   // Publish the joy message
   useEffect(() => {
-
     if (!config.publishOn) {
       return;
     }
 
-    if (pubTopic && pubTopic === config.pubTopic) {
+    if (config.pubTopic) {
+      console.log("About to publish on " + config.pubTopic);
       context.publish?.(config.pubTopic, config.publishConverted ? convMsg : rawMsg);
     }
-  }, [
-    context,
-    config.pubTopic,
-    config.publishOn,
-    config.publishConverted,
-    rawMsg,
-    convMsg,
-    pubTopic,
-  ]);
+  }, [context, config.pubTopic, config.publishOn, config.publishConverted, rawMsg, convMsg]);
 
   // invoke the done callback once the render is complete
   useEffect(() => {
@@ -202,4 +181,4 @@ function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Elem
 
 export function initExamplePanel(context: PanelExtensionContext): void {
   ReactDOM.render(<ExamplePanel context={context} />, context.panelElement);
-}
\ No newline at end of file
+}
```

jtbandes, 2023-03-02:
As for your comment about ROS 1 vs ROS 2 datatypes, I think that's just a quirk of the Rosbridge server / ROS 2 itself. The datatypes it reports in the topic list and the datatypes it accepts for advertisements may have different requirements or implementation details.

joshnewans, 2023-03-02:
Amazing, thanks! I'll try and give that PR a go tonight and let you know how I get on.

And yes I only had those other state variables in there as part of my experimenting to try and force the order - I'm happy to be rid of them!

Re the datatypes, that's really interesting, I didn't know that it was flexible even on the ROS end (although I've probably accidentally made use of that in the past without realising).

jtbandes, 2023-03-24:
Seems like it should be fine, but do we also need this in other players, e.g. FoxgloveWebSocketPlayer?

achim-k, 2023-03-24:
> Seems like it should be fine, but do we also need this in other players, e.g. FoxgloveWebSocketPlayer?

Only the rosbridge player publishes messages as JSON encoded string, so republishing should not be a problem for the other players (will confirm this)

jtbandes, 2023-03-24:
https://github.com/foxglove/studio/blob/41ece60bddc4e6768c1b0fc68370c30ec21200bb/packages/studio-base/src/players/FoxgloveWebSocketPlayer/index.ts#L829

achim-k, 2023-03-24:
> https://github.com/foxglove/studio/blob/41ece60bddc4e6768c1b0fc68370c30ec21200bb/packages/studio-base/src/players/FoxgloveWebSocketPlayer/index.ts#L829

Oh, you are right. I didn't think about JSON encoding :see_no_evil:

commit 732c6374 (2023-03-27): Encode typed arrays as arrays instead of objects when publishing messages as JSON (#5632)

**User-Facing Changes**
- The publish() API for extensions now supports publishing typed arrays

**Description**
This PR adds a custom JSON.stringify replacer function to encode typed
arrays as arrays rather than objects with numeric keys.

From
https://github.com/foxglove/studio/issues/5455#issuecomment-1483075718:

> Default behavior, without replacer:
> 
> ```js
> JSON.stringify({ val: new Uint8Array([1,2,3])});
> // Results in 
> // '{"val":{"0":1,"1":2,"2":3}}'
> ```
> 
> With replacer:
> 
> ```js
> [... cut]