foxglove/studio #5479: Extension panel sometimes receives message after unsubscribing

jtbandes, 2023-03-04:
**Description**
Extension panels sometimes receive messages in `currentFrame` _after_ a `context.subscribe([])` call has been made to unsubscribe from the topic.

**Steps To Reproduce**
Use this mcap file: https://github.com/foxglove/mcap/blob/fd3ecada218f41929c6d430acba3561e5ebfaf08/testdata/mcap/demo.mcap

Load this example panel extension:

```ts
import { PanelExtensionContext, RenderState } from "@foxglove/studio";
import { useLayoutEffect, useEffect, useState } from "react";
import ReactDOM from "react-dom";

const TOPIC = "/image_color/compressed";

function ExamplePanel({ context }: { context: PanelExtensionContext }): JSX.Element {
  const [renderDone, setRenderDone] = useState<(() => void) | undefined>();

  const [eventLog, setEventLog] = useState<string[]>([]);
  const [subscribeEnabled, setSubscribeEnabled] = useState(true);

  useLayoutEffect(() => {
    setEventLog((prev) => [...prev, "set up onRender listener"]);
    context.onRender = (renderState: RenderState, done) => {
      setRenderDone(() => done);
      setEventLog((prev) =>
        renderState.currentFrame
          ? [...prev, `currentFrame: ${renderState.currentFrame.length} messages`]
          : prev,
      );
    };
    context.watch("currentFrame");
  }, [context]);

  useEffect(() => {
    context.updatePanelSettingsEditor({
      nodes: {
        general: {
          fields: {
            subscribe: {
              input: "boolean",
              label: "Subscribe",
              value: subscribeEnabled,
            },
          },
        },
      },
      actionHandler(action) {
        if (action.action === "update" && action.payload.input === "boolean") {
          const on = action.payload.value ?? true;
          setEventLog((prev) => [
            ...prev,
            on ? "subscribe setting enabled" : "subscribe setting disabled",
          ]);
          setSubscribeEnabled(on);
        }
      },
    });
  }, [context, subscribeEnabled]);

  useEffect(() => {
    if (subscribeEnabled) {
      setEventLog((prev) => [...prev, `subscribe to ${TOPIC}`]);
      context.subscribe([TOPIC]);
      return () => {
        setEventLog((prev) => [...prev, `unsubscribe from ${TOPIC}`]);
        context.subscribe([]);
      };
    }
    return undefined;
  }, [context, subscribeEnabled]);

  useEffect(() => {
    renderDone?.();
  }, [renderDone]);

  return (
    <div style={{ padding: "1rem", overflowY: "auto", maxHeight: "100%" }}>
      <pre>{eventLog.join("\n")}</pre>
    </div>
  );
}

export function initExamplePanel(context: PanelExtensionContext): () => void {
  ReactDOM.render(<ExamplePanel context={context} />, context.panelElement);
  return () => {
    ReactDOM.unmountComponentAtNode(context.panelElement);
  };
}
```

Toggle the subscribe setting in the sidebar during playback. Observe that occasionally messages are received _after_ the subscribe setting has been turned off.

https://user-images.githubusercontent.com/14237/222863255-2cdeba4f-d82d-4bf4-b678-0f14679e3aca.mov

**Expected Behavior**
A message should not be received in `currentFrame` after `context.subscribe()` has been called to unsubscribe from the topic.

defunctzombie, 2023-03-19:
I've tried to reproduce this locally but am unsuccessful. I got it to happen _once_ on the production deployment but never again. I also tried writing some tests around this behavior both for renderState and for PanelExtensionAdapter and so far have not been able to get tests to trigger the behavior.

Part of me suspects that the fix for https://github.com/foxglove/studio/pull/5543 might help this too however without a reliable way to reproduce I can't say for sure.

defunctzombie, 2023-03-23:
Notes from our debugging session:

We determined the issue is with the example panel code presented above. It calls a react setter for `setEventLog` and provides a function to the react useState setter. This setter function closes over the `renderState` from the onRender function but might not run _immediately_. The unsubscribe cleanup from the useEffect might run before this setter function. This gives the appearance that a renderState happened after unsubscribing when this is actually the renderState that was closed over.

```
setEventLog((prev) =>
        renderState.currentFrame
          ? [...prev, `currentFrame: ${renderState.currentFrame.length} messages`]
          : prev,
      );
```

The determination is that there's no bug in Studio that's exposed here.

closed 2023-03-23 by defunctzombie