foxglove/studio #5482: feat(build-perf): Explore SWC (without mangling idents)

jon-chuang, 2023-03-06:
>> Are we going back to unreadable stack traces in prod? Is there any equivalent setting in swc?
>
> There is no equivalent setting in SWC.

This is, at least as of today, false. See `keepClassNames`, `keepFnNames`, `keepPrivateProps` in https://swc.rs/docs/configuration/minification.

I think we can explore using SWC-based build again if it makes the build faster.

_Originally posted by @jon-chuang in https://github.com/foxglove/studio/issues/3860#issuecomment-1455288294_

defunctzombie, 2023-03-06:
👍 sounds good - even keeping function names and class would be a win without private props

jon-chuang, 2023-03-06:
As it turns out, esbuild is as of today faster than terser: https://github.com/privatenumber/minification-benchmarks

defunctzombie, 2023-03-06:
Do you have a PR for studio to reproduce these results?

jtbandes, 2023-04-21:
Hi @jon-chuang – any update on this effort? Just wondering if we should keep this ticket open or close it out if there's no more work planned.

jtbandes, 2023-04-27:
Closing for now but feel free to let us know if you have any future explorations. Please note that we've recently switched our architecture so studio.foxglove.dev is built with some extra closed-source features, and as such we do not intend to provide public source maps for the prod builds anymore.

closed 2023-04-27 by jtbandes