foxglove/studio #5484: Switching the layout will cause the reset view to be in a different position

Woodii1998, 2023-03-06:
**Description**
Switching the layout will cause the reset view to be in a different position

https://user-images.githubusercontent.com/33593572/223032121-3e2cb720-ab45-4622-b02c-08dec274b17d.mp4

layout: 
https://drive.google.com/file/d/1WGfS87CKpwmH5ER3e8I4uLwvjV6qa19z/view?usp=sharing
https://drive.google.com/file/d/1pz_9Dtlzh0JHLKDtwtMnr6K5HNHkFvpW/view?usp=sharing

bag file:
https://drive.google.com/file/d/1FC5J5EucTOQ35G0J1-UdhWQVL0JXP_Bu/view?usp=sharing

- Version: online version
- Platform: web

**Steps To Reproduce**

Like in the video

ps: Caching will affect the reproduction of the problem, to ensure that you can reproduce it every time you can use chrome's empty cache and hard reload like in the video

**Expected Behavior**
Consistent effect before and after switching layout

foxymiles, 2023-03-21:
It looks to me like something has gone wrong earlier in this flow. Your initial view is this:

[image]

But when you reload you have a different view like this:
[image]

Even though the camera settings are the same.

foxymiles, 2023-03-21:
Is it possible for you to privately share the layouts involved here?

libertychisasuro, 2023-03-21:
@Woodii1998 looks like the link to your bag file is invalid

foxymiles, 2023-03-21:
> Is it possible for you to privately share the layouts involved here?

Oh nevermind I see you've attached them I'll take a look.

Woodii1998, 2023-03-21:
> @Woodii1998 looks like the link to your bag file is invalid

I seem to have deleted my files by mistake, you can try with this link

https://drive.google.com/file/d/1FC5J5EucTOQ35G0J1-UdhWQVL0JXP_Bu/view?usp=sharing

Woodii1998, 2023-03-21:
> > Is it possible for you to privately share the layouts involved here?
> 
> Oh nevermind I see you've attached them I'll take a look.

@foxymiles Not every bag can reproduce this problem, you can try this bag that I provide

foxymiles, 2023-03-21:
Thanks for the extra detail @Woodii1998.

I'm trying to reproduce here with the following steps:
1. Load your bag and "Test Layout 2".
2. Do "empty cache and hard reload"
3. Reset the view

In your video this seems to center the view on the base_odom object, but in my case it centers it on the purple sphere. Can you confirm that doing those three steps alone centers the view on base_odom?

And can you provide some extra detail on your environment please? Chrome version and OS?

Woodii1998, 2023-03-21:
> Thanks for the extra detail @Woodii1998.
>
> I'm trying to reproduce here with the following steps:
>
> 1. Load your bag and "Test Layout 2".
> 2. Do "empty cache and hard reload"
> 3. Reset the view
>
> In your [...]

Yes, with these three steps I can still reproduce the problem consistently

My Chrome version and OS：

[image]
[image]

Woodii1998, 2023-03-21:
@foxymiles You can follow my steps exactly and try again, all my friends around me can reproduce the problem steadily

https://user-images.githubusercontent.com/33593572/226526966-289e39ac-1e0d-423f-8bd8-b579d595507b.mp4

foxymiles, 2023-03-21:
@Woodii1998 Ok I can see a difference in behavior here if I load the bag directly vs uploading it to data platform first and them streaming it from there. That helps. Thanks for the extra detail.

foxymiles, 2023-03-21:
Ok I think I found an even simpler repro. Just repeatedly refreshing the page and waiting for it to render gives different views on some loads. I think this might have something to do with TF loading. Investigating.

libertychisasuro, 2023-03-21:
I am able to repro on macOS Monterey 12.6 (21G115) with Chrome 111.0.5563.64 (Official Build) (arm64).

@foxymiles would it be correct to expect `config.scene.syncCamera` to resolve to true in `ThreeDeeRender.tsx` such that the value of `cameraState` would be the exact same for both templates when it is saved in IndexexDB ?

libertychisasuro, 2023-03-21:
@foxymiles please disregard my previous comment. I just checked out the second video and I cannot repro as of yet.

snosenzo, 2023-03-29:
Able to repro, don't need to clear cache or anything, the most important aspect is that the camera follow modes are Fixed Frame and time starts at the _very_ beginning where the display frame transform isn't present yet. 

Steps:
 1. Have 2 layouts with a non-Pose follow mode saved with a 3D panel and a data platform source that doesn't have transforms at the beginning.
 2. Reload the source with time set to the very beginning, the 3D panel won't be showing most data because its at the beginning
 3. Hit play and let playback come in for a few seconds. See that data comes in but the camera center stays on the root frame. This is the issue because the fixed frame pose snapshot was initiated at the root frame and not the display frame because it hadn't come in yet. So all fixed frame camera settings are relative to that initial root frame instead of the display frame. Resetting the camera here will center the camera on the root frame
 4. Switch layouts. The camera in the fixed follow mode will be initiated at the display frame because the display frame now exists at the current time. Resetting the camera will center on the display frame (as intended).
 5. If you switch back to the original layout at the same timestamp it will act normally with the camera resetting to the display frame.

Unfortunately we can't always guarantee that resetting the camera with a fixed frame will always center on the same place because when the non-pose follow mode is initiated we take a snapshot of the post at the current time and use that as the offset for rotation and translation in `Fixed` and `Position` follow modes. We do this so that resetting the camera in the `Fixed` camera mode doesn't potentially reset you all the way back to 0, 0, 0 at your root frame which can be pretty jarring if the root frame is very far away.
 
However I will introduce a fix such that we don't set that pose snapshot until the display frame is present, which should at least initiate the fixed camera pose at the first display frame rather than the root frame. That should solve this issue when starting from a time where the display frame isn't present going into a time when it is

foxymiles, 2023-04-03:
@snosenzo Can we close this now?

defunctzombie, 2023-04-06:
This was fixed in https://github.com/foxglove/studio/pull/5662

closed 2023-04-06 by defunctzombie