foxglove/studio #5485: Entities missing and overlapping boxes in SceneUpdate topic in foxglove studio

pezy, 2023-03-06:
**Description**

I have encountered a bug in foxglove studio where the number of entities in the box_array/0 (foxglove.SceneUpdate) topic may change with timestamp, and this causes the boxes located near the root position to overlap. When I select these boxes, I found that their IDs are not in the entities list at the current timestamp (checked using the Raw Messages plugin). However, when I press <- to go back, the boxes immediately return to normal.

I am wondering if this is a bug or if there is an error in the messages I am sending.

- Version: 1.42.1
- Platform: Windows

**Steps To Reproduce**

[bug_demo.zip](https://github.com/foxglove/studio/files/10895060/bug_demo.zip)

https://user-images.githubusercontent.com/1147451/223037117-58b96a2b-e106-4fb2-a628-4939fdbeaeec.mp4

**Expected Behavior**

solution to avoid the overlapping of boxes and ensure all entity IDs are available.

snosenzo, 2023-03-16:
So I believe it's an issue with how you've constructed your `SceneUpdate`.
Stepping through the file: 
 - First `SceneUpdate`: the center box is in entity with id: 9  
 - Second `SceneUpdate`: there is another new center box with `id: 4` along with the existing `id: 9` box. Entity with `id: 9` is not present in the update and thus will not be removed from the scene. In order to remove it from the scene there must be either a deletion entry on the scene update or there needs to be an empty entity with `id: 9`  
 - Third SceneUpdate: Box from frame `id: 9` is still present and there is a  new box at the center of the frame with `id: 5`. The box with `id: 4` was moved elsewhere in the scene.  

The entities within SceneUpdates should represent the same objects across time so that when a new entity with the same id is received it will replace the old entity. So either you can add all of your boxes to the same SceneEntity on the SceneUpdate which will replace all old boxes with new ones, or if you want to track them across different SceneEntities, you'll need to make sure that each SceneEntity represents the same object/objects across time.  

Hope this helps!

pezy, 2023-03-20:
Thank you for your detailed explanation. I believe this should be due to a misuse of SceneUpdate. Therefore, it is not a bug but rather designed in this way. Thank you again~

closed 2023-03-20 by defunctzombie