foxglove/studio #5544: occupancy grid data has different performance in 3D pannel settings

YunxianHua, 2023-03-14:
**Description**
[image]

- Version: 1.43.0 and online version
- Platform: ubuntu 20.04

**Steps To Reproduce**

I selected `Custom` for `Color mode` (occupancyGrid) in previous foxglove studio version. with the update of foxglove studio, the performance of `Custom` like the picture below
[image]

I select the `color mode` to `map`, then select to `Custom`, it will have the right performance.

**Expected Behavior**

defunctzombie, 2023-03-16:
What are the types of `/map` and `/elevator_map` ?

jtbandes, 2023-03-16:
When I select Custom for an OccupancyGrid topic I see the min/max/unknown/invalid color settings as shown:

https://user-images.githubusercontent.com/14237/225714420-cc04a758-6fdc-4080-b2bb-031c2d71feb7.mov

YunxianHua, 2023-03-18:
> What are the types of `/map` and `/elevator_map` ?

it's occupancy grid data

YunxianHua, 2023-03-18:
> When I select Custom for an OccupancyGrid topic I see the min/max/unknown/invalid color settings as shown:

it happened when I upgrade the version of foxglove studio.
before upgrade, I use `Custom` to display the `occupancyGrid` data. 
after upgrade, the `color mode` is `Custom`, but the UI is like this, also, alpha can not work.
[image]

after upgrade, select the `color mode` to `map`, and then select to `custom`, it will back to normal like this
[image]

defunctzombie, 2023-03-18:
> after upgrade, select the `color mode` to `map`, and then select to `custom`, it will back to normal like this

It sounds like this has fixed your problem and you are able to change the color ranges using the `Custom` mode?

commit 1e86a944 (2023-03-27): Fix occupancy grid default settings in 3d panel (#5630)

**User-Facing Changes**
Fix occupancy grid default settings in 3d panel.

**Description**
Fix occupancy grid default settings in 3d panel. We default the color
mode to `custom`, but we didn't take this default into account when
building the associated fields.

Fixes: #5544