foxglove/studio #5551: Foxglove Studio log messages do not persist

sharonlinto, 2023-03-15:
**Description**
In Foxglove Studio, I am trying to use the Log panel to display warnings/errors to the user. However, upon refreshing the page or resizing the panel, the messages disappear (and would randomly appear sometimes when refreshing). 

https://user-images.githubusercontent.com/30011109/225458282-b0c08b7b-4130-437d-ab0c-1bb9c06d7d6c.mov

\
**Steps To Reproduce**
Display messages in the [Log panel](https://github.com/foxglove/schemas/blob/main/schemas/proto/foxglove/Log.proto) and resize/refresh the page. 

**Expected Behavior**
Log messages should be persistent (since they are being read from MCAP file).

jtbandes, 2023-03-15:
Hi, thanks for the report. Would you be able to share the mcap file you're working with so we can reproduce the issue more easily?

sharonlinto, 2023-03-15:
@jtbandes Sure, do you have an email I could send it to?

jtbandes, 2023-03-15:
my first name at foxglove.dev :)

jtbandes, 2023-03-17:
This should be fixed now via #5567 & #5568. Thanks for reporting!

closed 2023-03-17 by jtbandes