foxglove/studio #5561: create_publisher() failed setting topic QoS,

sgf201, 2023-03-16:
**Description**
create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'
the topic is not advertised
- Version:ROS2 humble  bridge with WebSocket++/0.8.2
- Platform: ubuntu22

**Steps To Reproduce**

Qos set to rclcpp::QoS qos(rclcpp::KeepLast{7});

**Expected Behavior**
topic advertised
>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

  'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.102152168] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[WARN] [1678937020.102373126] [foxglove_bridge]: Ignoring client unadvertisement from 192.168.6.195:52050 for unknown channel 3, client has 0 advertised topic(s)
[INFO] [1678937020.104918817] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 4
[INFO] [1678937020.105093695] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 4
[INFO] [1678937020.109083689] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/move_base_simple/goal" (geometry_msgs/PoseStamped) on channel 5
[INFO] [1678937020.110877709] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/clicked_point" (geometry_msgs/PointStamped) on channel 6
[INFO] [1678937020.111063987] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /move_base_simple/goal (0 subscribers) on channel 5
[INFO] [1678937020.111169025] [foxglove_bridge]: Client 192.168.6.195:52050 is no longer advertising /clicked_point (0 subscribers) on channel 6

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

  'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.114041753] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116

>>> [rcutils|error_handling.c:108] rcutils_set_error_state()
This error state is being overwritten:

  'Unknown QoS history policy, at ./src/qos.cpp:61'

with this new error message:

  'create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226'

rcutils_reset_error() should be called after error handling to avoid this.
<<<
[ERROR] [1678937020.116081249] [foxglove_bridge]: Failed to create publisher: could not create publisher: create_publisher() failed setting topic QoS, at ./src/publisher.cpp:226, at ./src/rcl/publisher.c:116
[INFO] [1678937020.118165425] [foxglove_bridge]: Client 192.168.6.195:52050 is advertising "/initialpose" (geometry_msgs/PoseWithCovarianceStamped) on channel 9

sgf201, 2023-03-16:
> Linear: [FG-2507](https://linear.app/foxglove/issue/FG-2507/create-publisher-failed-setting-topic-qos-%5Bfoxglovestudio-5561%5D)

[image]

please give a ticket for the workspace

achim-k, 2023-03-16:
This seems to be a `foxglove_bridge` issue, not a Studio issue. Could you create an issue at https://github.com/foxglove/ros-foxglove-bridge including detailed reproduction steps? Thanks!

defunctzombie, 2023-03-17:
Closing this as a duplicate of https://github.com/foxglove/ros-foxglove-bridge/issues/200

closed 2023-03-17 by defunctzombie