foxglove/studio #5577: a very big value of uint64 / int64 fields in protobuf message will be parsed to a fault value.

lizhucheng, 2023-03-17:
**Description**
a uint64 field in a proto will be parsed to fault value when  the value is big;
e.g. 2394304762687298819 -> 2394304762687299000

- Version: all
- Platform: all system, browser

**Steps To Reproduce**
1. create a proto file  declare a uint64 field;
test.proto
 message Test {
   uint64 a = 1;
}
2. write a messge to a pb file, let the value of the field to be  2394304762687298819;
3. create a mcap file which contains pb data and schema above;
4. play the mcap file in foxglove studio.
5. check the message in a raw panel.
6. 2394304762687298819 -> 2394304762687299000

**Expected Behavior**
2394304762687298819 -> 2394304762687298819

[Related articles](https://github.com/protobufjs/protobuf.js/issues/1838)

defunctzombie, 2023-03-17:
Can you please provide a sample file where you experience this issue. That will help us make sure we are looking at the same type of data you are.

jtbandes, 2023-04-07:
Hi @lizhucheng, please try out this version of the app https://studio-j53ommwbi.foxglove.party/ (from #5727) and let me know if it works for you.

closed 2023-04-07 by jtbandes

lizhucheng, 2023-04-14:
> Hi @lizhucheng, please try out this version of the app https://studio-j53ommwbi.foxglove.party/ (from #5727) and let me know if it works for you.

Yes，it works for me. Good job and Many thanks.