foxglove/studio #5591: Support for Ubuntu 20.04

namrata-jangid, 2023-03-20:
Hi, 

Our team is having an issue with running Foxglove Studio on an Ubuntu 20.04 machine. We had opened an issue previously as well (https://github.com/foxglove/studio/issues/5564).

Attaching a few screenshots to further elaborate on the issue:

- Ubuntu version
[image]

- Installing Foxglove Studio:
[image]

- 'foxglove-studio' command does not run as expected:
[image]

Requesting your help on this. 

Thank you! :)

jtbandes, 2023-03-20:
We're going to need more information to be able to help. The app works fine with my ubuntu 20.04 VM on arm64. Do you get any error messages?

[image]

kis-balazs, 2023-03-20:
Hi!
Weird thing is exactly what you asked, there is nothing popping up as an error, and neither ```htop``` shows any significant move regarding a "foxglove-studio" executable (see last image)

We are running it on a Jetson Xavier, with JetPack 5.0.2, Ubuntu 20.04 as specified in the initial image. Can maybe this trigger such unusual behavior?

How could we figure this out better?

Thanks,
 Balazs

jtbandes, 2023-03-20:
Are you able to run other apps built with Electron? https://github.com/electron/electron  You could also try https://www.electronjs.org/fiddle/. It sounds like there are some more logs you can enable as well: https://www.electronjs.org/docs/latest/api/command-line-switches#--enable-loggingfile

kis-balazs, 2023-03-23:
Hi there!
We have run fiddle, and it has the same behavior, even though it shows up in the htop window, as being present... very weird behavior, and I believe it is an electron-related issue in this case. Do you have any idea how to solution this?
Please see screenshots below.

[image]
[image]
[image]
low.

jtbandes, 2023-03-23:
Sorry, I've never experienced this problem before. I don't have any particular advice other than to search through the electron community to see if others have experienced this problem. I'm going to close this issue because we've determined the bug is not related to Foxglove Studio, but feel free to post further updates and we can reopen it if anything changes.

closed 2023-03-23 by jtbandes

kis-balazs, 2023-03-23:
Yeah well this wasn't useful. Then please take out the arm64 support from the foxglove website until the error gets solved, since you are in fact, not supporting arm64 architecture fully :)

jtbandes, 2023-03-23:
Did you try running with `--enable-logging`? Any interesting log messages? https://www.electronjs.org/docs/latest/api/command-line-switches#--enable-loggingfile