foxglove/studio #5592: Interactive operation: Page turning conflicts with scroll wheel zoom

smbera, 2023-03-20:
**Description**
before repair：

https://user-images.githubusercontent.com/23401125/226315547-cae3b374-0096-4727-a4cd-c96dcefe4268.mp4

From this video, it can be seen that when I use the mouse wheel to scroll down and turn the page, the X-axis will shrink accordingly. This is because the operation of scrolling the mouse wheel down to turn the page conflicts with the operation of zooming the chart.

after repair：

https://user-images.githubusercontent.com/23401125/226315608-9621cbbf-a7fc-47f5-8aeb-b5bb36391d9d.mp4

Here, I use "Ctrl" + mouse wheel to zoom the chart, but just use the mouse wheel to scroll down and turn pages.

- Version: 1.30.0-dev
- Platform: web

**Steps To Reproduce**

**Expected Behavior**

smbera, 2023-03-21:
> The current version of the app is 1.44.0. Please try with the latest version. Also, please provide more information about what platform/browser you are using. I am unable to reproduce this issue on my system.

ok, I update the information

Version: 1.44.0-dev
Platform: Chrome 版本 110.0.5481.100（正式版本） （64 位）

jtbandes, 2023-03-22:
OK, I understand now that you are talking about the conflict between scrolling the state transitions list (1) and zooming (2). I was confused because your video also shows scrolling the whole page (3) which is unexpected.
[image]

We have some work in progress (https://github.com/foxglove/studio/pull/5604) that will change the State Transitions panel a bit. However, I think this issue may still be relevant after that change.

PR #5678 (2023-04-03): Disable wheel events on state transition chart wrapper

**User-Facing Changes**
Fix zoom and scroll behavior for StateTransition panel when there are series outside of the visible panel area.

**Description**
See code comment for description.

Fixes: #5592
commit 1310c572 (2023-04-03): Disable wheel events on state transition chart wrapper (#5678)

**User-Facing Changes**
Fix zoom and scroll behavior for StateTransition panel when there are
series outside of the visible panel area.

**Description**
See code comment for description.

Fixes: #5592
closed 2023-04-03 by defunctzombie