foxglove/studio #5627: Unable to launch desktop app from terminal

kay-000, 2023-03-26:
I tried to follow the instructions from [Contributing to Foxglove Studio](https://github.com/foxglove/studio/blob/main/CONTRIBUTING.md)

but after trying " yarn desktop:serve " 

I get 22 errors in the terminal and I am not sure how to fix this issue. 

```
ERROR in ./packages/studio-base/src/App.tsx:125:12
TS2786: 'ErrorBoundary' cannot be used as a JSX component.
  Its instance type 'ErrorBoundary' is not a valid JSX element.
    123 |         {enableGlobalCss && <GlobalCss />}
    124 |         <CssBaseline>
  > 125 |           <ErrorBoundary>
        |            ^^^^^^^^^^^^^
    126 |             <MaybeLaunchPreference>
    127 |               <MultiProvider providers={providers}>
    128 |                 <DocumentTitleAdapter />

ERROR in ./packages/studio-base/src/components/Autocomplete.tsx:416:6
TS2786: 'ReactAutocomplete' cannot be used as a JSX component.
  Its instance type 'Autocomplete' is not a valid JSX element.
    The types returned by 'render()' are incompatible between these types.
      Type 'import("/Users/kaylamcfarlane/Desktop/workspace/foxglove-studio/node_modules/@types/react/index").ReactNode' is not assignable to type 'import("/Users/kaylamcfarlane/node_modules/@types/react/index").ReactNode'.
    414 |
    415 |   return (
  > 416 |     <ReactAutocomplete
        |      ^^^^^^^^^^^^^^^^^
    417 |       open={open}
    418 |       items={autocompleteItems}
    419 |       getItemValue={(item: FzfResultItem<T>) => getItemValue(item.item)}

ERROR in ./packages/studio-base/src/components/Panel.tsx:510:34
TS2786: 'PanelComponent' cannot be used as a JSX component.
  Its element type 'Component<unknown, unknown, any> | ReactElement<unknown, string | JSXElementConstructor<any>> | null' is not a valid JSX element.
    Type 'Component<unknown, unknown, any>' is not assignable to type 'Element | ElementClass | null'.
      Type 'Component<unknown, unknown, any>' is not assignable to type 'ElementClass'.
        The types returned by 'render()' are incompatible between these types.
          Type 'import("/Users/kaylamcfarlane/Desktop/workspace/foxglove-studio/node_modules/@types/react/index").ReactNode' is not assignable to type 'import("/Users/kaylamcfarlane/node_modules/@types/react/index").ReactNode'.
            Type '{}' is not assignable to type 'ReactNode'.
    508 |       [otherPanelProps, panelComponentConfig, saveConfig],
    509 |     );
  > 510 |     const child = useMemo(() => <PanelComponent {...childProps} />, [childProps]);
        |                                  ^^^^^^^^^^^^^^
    511 |
    512 |     const renderCount = useRef(0);
    513 |
```

foxymiles, 2023-03-27:
Please try running these commands then try running `desktop:serve` again.

```sh
rm -rf ./node_modules
yarn cache clean
yarn
```

kay-000, 2023-03-28:
This did not work, I received the same 22 errors.

foxymiles, 2023-03-28:
Can you provide some extra information about your environment please? Operating system & node version?

kay-000, 2023-03-28:
My operating system is Mac OS and I have v18.15.0 of node.

foxymiles, 2023-03-29:
Thanks for the extra details. I just did a fresh checkout of the repo and was able to build and run without issues. Could you please try deleting your entire `studio` folder, doing a fresh checkout from github and trying these commands:

```sh
yarn
yarn desktop:serve
```

kay-000, 2023-03-29:
I deleted the old folder and cloned a new one, but after trying the second command I received this error:
```
> yarn desktop:serve
yarn run v1.22.19
warning package.json: No license field
error Command "desktop:serve" not found.
info Visit https://yarnpkg.com/en/docs/cli/run for documentation about this command.
```

foxymiles, 2023-03-29:
Did you run the `corepack enable` command? Did it complete without errors?

jtbandes, 2023-03-29:
```
warning package.json: No license field
error Command "desktop:serve" not found.
```

Both of these errors indicate the command is not being run in the right directory or context. You can see a `license` field and a `desktop:serve` command are both present in this package.json file: https://github.com/foxglove/studio/blob/main/package.json

defunctzombie, 2023-04-01:
@kay-000 I've tested the latest version of Studio using node v18 on my mac and did not run into these issues.

Looking at one of your error messages another thing to try is removing this `node_modules` folder in your home directory.
```
/Users/kaylamcfarlane/node_modules/@types/react/index
```

I believe the issue is with your environment and where you are running the commands so I'm going to close this bug. If you continue to struggle getting this to work please join our [Slack](https://foxglove.dev/slack) and ask in #lounge and folks will be able to help you out.

closed 2023-04-01 by defunctzombie