foxglove/studio #5637: Image panel zoom-in with mouse scroll wheel is too coarse to be usable

jhurliman, 2023-03-27:
**Description**
When zooming in with my mouse scrollwheel, a single click (one wheel discrete movement of the wheel) causes the zoom percent to go from 50% to 558%, then to 6224%.

- Version: `main`
- Platform: Linux Pop OS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): MCAP

**Steps To Reproduce**
Use a mouse with a scroll wheel

**Expected Behavior**
Significantly smaller zoom steps per wheel position

defunctzombie, 2023-04-06:
I did some research into this issue. The tl;dr; is that the [wheel event](https://developer.mozilla.org/en-US/docs/Web/API/Element/wheel_event) I am seeing when using a scroll wheel mouse on Linux (Ubuntu for me) is sending large deltaY values (order of 120, 30, 60). These values for each "wheel" event are far greater than what I experience when using a touchpad. With a touchpad I see values in the 2-15 range (higher when I scroll faster). Indeed on linux (maybe not all linux?) zooming with the wheel is basically impossible.

This deltaY value is fed into the zoom logic via this module (https://www.npmjs.com/package/use-pan-and-zoom) and influences how much to zoom. When the deltaY values are so large the zoom amount is exaggerated as you are experiencing with your mouse. From the perspective of the zoom logic, it is being told the mouse wheel event has a large deltaY. Looking around the internet, it seems that this is possibly expected on Linux (tho is maybe browser specific?). Each wheel "tick" on linux produces this large deltaY. This seems to be different on MacOS and even Windows.

Since we are actively working on moving to a new architecture for the image panel, we are going to hold off on any hacks or workarounds for this bug for a few weeks as we prepare the new panel. I think it will use a different set of logic for zooming which might address the concerns raised here. If we still experience unusable zoom with the new panel we will revisit and make the zoom usable.

jtbandes, 2023-05-05:
Give this a try: https://github.com/foxglove/studio/pull/5915 https://studio-4z5aix4qe.foxglove.party/ 
I've added some clamping to the wheel events that I think strikes a decent balance where it feels fairly smooth on a macOS trackpad and is still usable on Linux with a wheel mouse.

defunctzombie, 2023-05-07:
I've tried the new image panel with my Ubuntu 22 machine and a physical mouse wheel and the behavior is much improved. I am able to use one wheel tick and zoom slightly into the image rather than some crazy several hundred % jump.

I also tried the zoom behavior with my Mac trackpad on osx and the behavior feels good there as well.

I am gonna close this bug as not-planned since we are on the eve of switching over to the beta image panel and won't be resolving this issue in the old image panel.

closed 2023-05-07 by defunctzombie