foxglove/studio #5643: Chart dithering

Woodii1998, 2023-03-28:
**Description**
When the chart panel is resized to a specific size, the chart will jitter

- Version: Latest
- Platform: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file

**Steps To Reproduce**

https://user-images.githubusercontent.com/33593572/228175799-7e110f00-6453-407a-8a9e-8de88a58db03.mp4

bag File: 
https://drive.google.com/file/d/1YVtcnAcYCDRrH5TsEjwReQspNiBwm1ZU/view?usp=sharing

layout
https://drive.google.com/file/d/10Sjr0djPfFTcwyF9ZIdMvtRRL5kcZsvs/view?usp=sharing

If you use a 16-inch macbook pro you can open this data using the layout I gave to reproduce, if it is other devices, then you may need to simply adjust the width and height of the chart

You can resize to the size in the picture
[image]

**Expected Behavior**
Chart Normal Display

Woodii1998, 2023-03-28:
https://github.com/foxglove/studio/blob/febaf50e4bfb61c61f4f001451649fdb212c613f/packages/studio-base/src/panels/Plot/index.tsx#L573

It seems to be because the minYValue and maxYValue are undefind when calling PlotPanel, and then chart.js has some problems when calculating the y-axis

Woodii1998, 2023-03-28:
https://github.com/Woodii1998/studio/pull/2/files 

I tried to fix it, and it seems to be possible to fix the problem by calculating the range of the y-axis outside, but of course this method may not be a good solution

foxymiles, 2023-03-30:
Thanks for the detailed bug report! I can reproduce it here.

foxymiles, 2023-04-03:
Closed by https://github.com/foxglove/studio/pull/5665

closed 2023-04-03 by foxymiles