foxglove/studio #5725: images in 3d only show the first frame and don't update

jtbandes, 2023-04-06:
`_recomputeCameraModel` returns undefined if the camera model did not change between the current frame and previous, but undefined `cameraModel` return value causes the call to `this._updateImageRenderable(renderable, image, cameraModel, receiveTime, settings);` in `handleImage()` to be skipped so the new image data is discarded

[Slack Message](https://foxglove.slack.com/archives/C028UEY858S/p1680814696349569)

jhurliman, 2023-04-06:
This should be the fix (in handleImage):

`const cameraModel = this._recomputeCameraModel(renderable, cameraInfo) ?? renderable.userData.cameraModel;`

jhurliman, 2023-04-06:
It's desirable for _recomputeCameraModel to return undefined if nothing changed so the handleCameraInfo callback can be a mostly no-op if the camera model didn't change between messages, and it's uncommon for the camera model to change in real-time. That's why I think fixing it in this one place is better than changing _recomputeCameraModel

defunctzombie, 2023-04-06:
> It's desirable for _recomputeCameraModel to return undefined if nothing changed so the handleCameraInfo callback can be a mostly no-op if the camera model didn't change between messages

👍 Thanks for that extra context. I see that the function has that comment that it returns undefined but now I understand why that is the case and how it impacts the logic.

PR #5728 (2023-04-07): Fix image rendering in 3d panel

**User-Facing Changes**
Fix image rendering in the 3d panel.

**Description**
This change resolves a regression introduces in https://github.com/foxglove/studio/pull/5555. The change in that PR incorrectly skipped updating the image when the _recomputeCameraModel returned undefined - thinking that it would always return the latest camera model.

This fix modes the _recomputeCameraModel logic into _updateImageRenderable and changes the inputs to _updateImageRenderable to accept the CameraInfo rather than the camera model. It also updates _recomputeCameraModel to [... cut]
commit 45f1d69c (2023-04-07): Fix image rendering in 3d panel (#5728)

This change resolves a regression introduces in
https://github.com/foxglove/studio/pull/5555. The change in that PR
incorrectly skipped updating the image when the _recomputeCameraModel
returned undefined - thinking that it would always return the latest
camera model.

This fix modes the _recomputeCameraModel logic into
_updateImageRenderable and changes the inputs to _updateImageRenderable
to accept the CameraInfo rather than the camera model. It also updates
_recomputeCameraModel to return the cached camera model if available.

_recomputeCameraModel w [... cut]
closed 2023-04-07 by defunctzombie