foxglove/studio #5736: Error playing point cloud data in 3D panel after opening up the Image panel

ShuhaoQing, 2023-04-10:
**Description**
I have built an mcap file using the nuScene data set. I have three topics, a tf, a point cloud and a compressed image. When I don’t have an image panel open, the point cloud is rendered correctly in the 3D panel. When I open the image panel and display the compressed image, the point clouds stop updating and the 3D panel updates at a lower rate. Does anyone know why this is? If it helps, this is an unindexed mcap file, running mcap cli recover on it fixes the problem. See the attached recording.

- Version: `main`
- Platform: macOS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): local unindexed mcap file

**Steps To Reproduce**

1. Open the following unindexed macp file. (The data got down-sampled due to github upload file size restriction)
[nuScene_low_quality.mcap.zip](https://github.com/foxglove/studio/files/11188331/nuScene_low_quality.mcap.zip)

2. Open up 3D panel only. The point cloud and tf data plays normally.

3. Open up both 3D panel and Image panel. The compressed image plays normally but point cloud stops updating, and tf updates much less frequently.

See the attached recording for reproducing the issue.

https://user-images.githubusercontent.com/115679637/230842730-058d2558-2c79-4f20-a4f4-eb808c216306.mp4

snosenzo, 2023-04-18:
Thanks for filing!
Fixed in this PR: https://github.com/foxglove/studio/pull/5787
Performance is expected to be degraded for unindexed mcap files, but this should fix messages on subscribed topics not going through.

closed 2023-04-18 by snosenzo

geoffliu, 2023-04-20:
@snosenzo Thank you! Can always count on FG team for a quick turnaround