foxglove/studio #5737: Line jittering in 3D panel

IvDmNe, 2023-04-10:
Hello. I try to visualize a map in 3D panel. For this I create LinePrimitive objects and send them in SceneUpdate message through websocket in protobuf format. In Studio these lines continuously jitter. And the lines supposed to be straight. I visualized them in  different 3D visualizers.

Below I attach screenshots of the same lines but with slightly different camera orientation.

[image]
[image]

- System: Window 11
- Data source : Foxglove Websocket connection

This doesn't happen with sample data.
Also, I want to mention that the coordinates are in UTM and x is about 3 * 10^5 and y - 6 * 10^6. Can such big numbers lead to float rounding with loss of precision?

Have anyone seen this bug before?

jtbandes, 2023-04-11:
> Can such big numbers lead to float rounding with loss of precision?

Yes, definitely.

Can you provide some steps to reproduce and example data so we can observe the issue?

I would also recommend you try changing the Near and Far settings under View options in the 3D panel:

[image]

amacneil, 2023-04-11:
Sounds like you are bumping up against the limits of float32 precision. Here's a [similar issue](https://github.com/foxglove/studio/issues/1738).

You can work around this by creating an intermediary frame using [FrameTransform](https://foxglove.dev/docs/studio/messages/frame-transform), then specifying your LinePrimitive objects relative to the local frame.

jtbandes, 2023-04-18:
Hi @IvDmNe, have you had a chance to try out some of the ideas mentioned above?

IvDmNe, 2023-04-19:
Hello, @jtbandes . Today I just confirmed that if I translate the map (x is about 3e4 and y - 1e5) it is being rendered without artifacts.

closed 2023-04-19 by IvDmNe

IvDmNe, 2023-04-26:
Also I want to mention that the similar effect happens when visualizing a path in 3D panel. Here I have two trajectories, but one of them is visualized as "Arrows" and the other as "Line". Both are without intermediate TF.

[image]

tlobig, 2024-01-08:
> Sounds like you are bumping up against the limits of float32 precision. Here's a [similar issue](https://github.com/foxglove/studio/issues/1738).
>
> You can work around this by creating an [...]

Sorry to unearth this old thing, but anyways where do you send the FoxgloveTransform messages to? i.e. what is the topic for it. With standard tf2 messages it's just /tf and Foxglove Studio picks them up organically. Are these custom transform messages needed at all? Does it magically center the data points at the provided translation during rendering? I tried to sift through available documentation and even some code, but since there is really a lot going on in Foxglove, I have a hard time figuring this out.

amacneil, 2024-01-08:
If you are using ROS, you can publish tf2 message on `/tf` and everything will work fine. You only need the Foxglove FrameTransform message if you are not using ROS (and I don't believe we require a specific topic name for those messages).

Our supported transform messages are listed [here](https://docs.foxglove.dev/docs/visualization/panels/3d/#transforms).