foxglove/studio #5754: Incorrect data source frequency statistics during real-time ROS1 connection packet playback

happykongmeng, 2023-04-12:
**Description**
Incorrect data source frequency statistics during real-time ROS1 connection packet playback

- Version: 1.39.0
- Platform: windows
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
-nuScenes-v1.0-mini-scene-0061.bag

**Steps To Reproduce**

1. on linux: rosbag play --clock --loop nuScenes-v1.0-mini-scene-0061.bag
2.on windows: ROS1 connection with linux and open Data source Info panel
3. data source frequency statistics keeps increasing，this is not correct.

**Expected Behavior**

achim-k, 2023-04-13:
From the issue description it's not clear what data source you are using. Do you play back a bag file and connect wtih Studio using one of the live connections data sources (rosbridge, ros 1/2 native, foxglove bridge)? Or, do you open the bag directly in studio?

happykongmeng, 2023-04-19:
1、You can rosbag play any rosbag and ros1 connection with the roscore, the frequency statistics is incorrect. 
2、Open the bag directly in studio，the frequency statistics is correct.

defunctzombie, 2023-04-24:
@happykongmeng Please upgrade to the latest version of studio. 1.39 is a few months old by now. Please report if you are still seeing this issue.