foxglove/studio #5755: [3D] The default scale of foxglove.PosesInFrame is shown as `3`, but is actually much smaller

jhurliman, 2023-04-12:
**Description**

Open an MCAP with a `foxglove.PosesInFrame` topic and visualize in the 3D panel. The settings sidebar shows it has a scale of `3`, but the actual rendered size is much smaller (1?). Type in `3` and you can see the scale increase. Remove the typed in `3`, replacing it with a default value `3` and observe the scale decrease.

foxymiles, 2023-04-24:
Do you have a sample mcap that demonstrates this? When I use nuscenes and a converter screen from `/pose` the default scale is 0.2 which seems to correspond with what's actually being rendered.

[image]

snosenzo, 2023-04-25:
@foxymiles John mentioned the issue is with `foxglove.PosesInFrame` not `foxglove.PoseInFrame` (singular). It looks like your convert script outputs a `PoseInFrame`, which would be why your repro isn't showing the issue.

closed 2023-04-25 by snosenzo

foxymiles, 2023-04-25:
> @foxymiles John mentioned the issue is with `foxglove.PosesInFrame` not `foxglove.PoseInFrame` (singular). It looks like your convert script outputs a `PoseInFrame`, which would be why your repro isn't showing the issue.

Thanks I misread that.

PR #5826 (2023-04-25): Fix incorrect placeholder text for `PosesInFrame` Axis Scale
**User-Facing Changes**

 - Fix incorrect placeholder text for `PosesInFrame` Axis Scale

**Description**
Incorrect variable used for axis scale

Fixes: #5755
FG-2911
commit 24edd05a (2023-04-25): Fix incorrect placeholder text for `PosesInFrame` Axis Scale (#5826)

**User-Facing Changes**

 - Fix incorrect placeholder text for `PosesInFrame` Axis Scale

**Description**
Incorrect variable used for axis scale

Fixes: #5755
FG-2911
commit c6ea5ec5 (2023-04-26): Fix incorrect placeholder text for `PosesInFrame` Axis Scale (#5826)

**User-Facing Changes**

 - Fix incorrect placeholder text for `PosesInFrame` Axis Scale

**Description**
Incorrect variable used for axis scale

Fixes: #5755
FG-2911