foxglove/studio #5756: Resizing custom panel reinitializes the panel

chfritz, 2023-04-12:
**Description**

- Version: 1.39.1
- Platform: ubuntu 20.04, chrome
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): bag file

**Steps To Reproduce**

https://user-images.githubusercontent.com/4239098/231558041-ecacf032-0f94-4325-9812-16ee5393f271.mp4

[foxglove-test-panel.zip](https://github.com/foxglove/studio/files/11214960/foxglove-test-panel.zip)

Watch the video, then, to reproduce, follow these steps:
- Open foxglove in a browser
- open a bag file with image topics
- add the test panel by drag&dropping the foxe file in the zip into the browser
- add the panel
- open the panel settings and select a ros topic
- resize the panel
- see the settings reset to

What's bizarre is that this only seems to happen the first time the panel is resized, not afterwards. Also changing the panel settings once more doesn't seem to re-trigger this issue.

**Expected Behavior**

Behave like it does on subsequent resizes, i.e., no reset and keep the settings.

chfritz, 2023-04-12:
This actually doesn't seem to depend on the custom panel. The plot panel seems to re-initialize as well on the first resizing, as seen in the console in this video without custom panels.

https://user-images.githubusercontent.com/4239098/231562219-48385eb9-818c-4b1f-be06-9113b82da2e5.mp4

jtbandes, 2023-04-12:
Thought this was fixed in https://github.com/foxglove/studio/pull/5567, but guess not...

chfritz, 2023-04-12:
@jtbandes Thanks for the pointer. Let me try on the latest. I was using v1.39.1 and didn't notice there were newer versions (my bad).

chfritz, 2023-04-12:
Confirmed, it's working correctly in v1.50. Sorry for the confusion and thanks again.

closed 2023-04-12 by chfritz