foxglove/studio #5795: 3D Panel Transform Error

oosthoek, 2023-04-19:
**Description**

When displaying PCL in its ego frame, It it still being affected by other transforms in the TF tree.

- Version: 1.51.0+2a2c721.4dbf99c
- Platform: web
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbridge

**Steps To Reproduce**

- play bag with pcl and tf topics
- configure 3D panel to display "luminar_lidar_0" frame

https://user-images.githubusercontent.com/6978789/233157808-acd0ff50-2baf-4384-9732-985e82683e64.mp4

[pcl_only2.zip](https://github.com/foxglove/studio/files/11276294/pcl_only2.zip)
[pcl_tf2.zip](https://github.com/foxglove/studio/files/11276295/pcl_tf2.zip)

**Expected Behavior**
pcl renders untransformed

oosthoek, 2023-05-01:
Hello.  Wondering if anyones had the time to look into this issue yet?

jtbandes, 2023-05-01:
Not yet, but we have it on our list to get to sometime this week.

snosenzo, 2023-05-04:
Thanks for filing! After looking into it, it appears that studio is showing your data properly and there is something erroneous in your base_link transform data. If you plot the base_link position.y over time it appears to have spikes causing the display frame to jump around.

https://user-images.githubusercontent.com/10187776/236241558-a98a1624-871f-4539-8696-4ffd61d1e6c1.mov

I can't say I know why RViz would be displaying this without those jumps.

oosthoek, 2023-05-04:
yes i understand the odom is erroroneous. but if the pcl is in frame luminar_lidar_0 and the "display frame"  is set to luminar_lidar_0 then why would the odom value in the /tf affect the visualization?

snosenzo, 2023-05-05:
Sorry for the confusion. To explain why the odom value is affecting the visualization: Right now, the transformation of 3D object (renderable) poses are always from renderable's frame (eg: in `header` of message) to the root frame at the header stamp time, then from the root frame to the display frame at the currentTime. If the renderable's timestamp is different from currentTime, we _need_ to  go through the root frame in order to make sure that the transform is where it should be because there’s a possibility that a frame between the renderable frame and root could have a different transform at that time. In this case the renderable headerstamp is defined and won’t be using the current time, so there’s not a way that we can perform this transform according to that headerstamp without going through the root frame. 
(to be clear in the above passage, the root frame is the fixed frame.)

However, in non-pointcloud 3D topic schema types there's a "Frame lock" setting that overrides the message header stamp such that the message is always positioned according to the transforms at the _current_ time (most recent version of the frame). Note that this does not display the data as accurately because those messages were created according to that older frame transform. 
This PR demonstrates the behavior of what this would look like if frame lock was turned on for point clouds: https://github.com/foxglove/studio/pull/5923

So to reiterate studio is working as intended, but would this feature be beneficial and cover you needs? or does this explanation suffice?

Let me know if anything is unclear or if you have any questions

snosenzo, 2023-05-12:
@oosthoek Since I there hasn't been a response, I'm going to close this issue for now since things are acting as expected from the studio perspective. Open to answer any more questions you might have about this behavior though and if you want to open a discussion to request frame locked pointclouds as a feature please do so!

closed 2023-05-12 by snosenzo