foxglove/studio #5818: 3D crashed

263868161, 2023-04-25:
**Description**
When I use 3D, a crash occurs.Works fine on older versions（1.2.0）.
[image]
[image]

- Version:1.51.0
- Platform:ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
/home/hy/temp/0424/parking_test/log/ora1_0423/0423_ora1_1.zip
**Steps To Reproduce**

**Expected Behavior**

jtbandes, 2023-04-25:
Please upload the `/home/hy/temp/0424/parking_test/log/ora1_0423/0423_ora1_1.zip` file for testing.

263868161, 2023-04-25:
[0423_ora1_1.zip](https://github.com/foxglove/studio/files/11315670/0423_ora1_1.zip)

jtbandes, 2023-04-25:
When I load this dataset, the 3D panel doesn't crash, but none of the topics are visualizable because there aren't any [types supported by the 3D panel](https://foxglove.dev/docs/studio/panels/3d#supported-messages). Is there more information about your setup that is missing from the bug report?

[image]
[image]

jtbandes, 2023-04-27:
I'm going to close this issue since we haven't heard back, but feel free to post below if you have more debugging information to share.

closed 2023-04-27 by jtbandes