foxglove/studio #5825: Somthing wrong in yarn install

WangShiJieAOTM, 2023-04-25:
**Description**
Somthing wrong in yarn install
- node Version:v16.20.0
- Linux Version:ubuntu18.04
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

$ yarn install
➤ YN0000: ┌ Resolution step
➤ YN0002: │ @foxglove/studio-base@workspace:packages/studio-base doesn't provide @babel/core (p883a7), requested by babel-jest
➤ YN0002: │ @foxglove/studio-base@workspace:packages/studio-base doesn't provide @babel/core (p90023), requested by babel-plugin-transform-import-meta
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides chart.js (p698a7) with version 4.2.1, which doesn't satisfy what @foxglove/chartjs-plugin-zoom requests
➤ YN0002: │ @foxglove/studio-base@workspace:packages/studio-base doesn't provide electron (p63772), requested by @foxglove/electron-socket
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides monaco-editor (p5138f) with version 0.30.1, which doesn't satisfy what react-monaco-editor requests
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides react (pbf2f7) with version 17.0.2, which doesn't satisfy what react-hover-observer requests
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides react (p476c7) with version 17.0.2, which doesn't satisfy what react-mosaic-component and some of its descendants request
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides react (p8d078) with version 17.0.2, which doesn't satisfy what react-virtualized requests
➤ YN0060: │ @foxglove/studio-base@workspace:packages/studio-base provides react-dom (p647bb) with version 17.0.2, which doesn't satisfy what react-virtualized requests
➤ YN0002: │ @foxglove/studio-base@workspace:packages/studio-base doesn't provide react-refresh (p63074), requested by react-refresh-typescript
➤ YN0002: │ @foxglove/studio-desktop@workspace:packages/studio-desktop doesn't provide @emotion/react (pc4a55), requested by tss-react
➤ YN0002: │ @foxglove/studio-desktop@workspace:packages/studio-desktop doesn't provide react-refresh (p6ca99), requested by @pmmmwh/react-refresh-webpack-plugin
➤ YN0002: │ @foxglove/studio-desktop@workspace:packages/studio-desktop doesn't provide react-refresh (p446d1), requested by react-refresh-typescript
➤ YN0002: │ @foxglove/studio-desktop@workspace:packages/studio-desktop doesn't provide typescript (pb34c7), requested by fork-ts-checker-webpack-plugin
➤ YN0002: │ @foxglove/studio-desktop@workspace:packages/studio-desktop doesn't provide typescript (pef492), requested by react-refresh-typescript
➤ YN0002: │ @foxglove/studio-web@workspace:packages/studio-web doesn't provide @emotion/react (p981ad), requested by tss-react
➤ YN0002: │ @foxglove/studio-web@workspace:packages/studio-web doesn't provide i18next (p0cdf7), requested by react-i18next
➤ YN0002: │ @foxglove/studio-web@workspace:packages/studio-web doesn't provide react-refresh (p92e70), requested by @pmmmwh/react-refresh-webpack-plugin
➤ YN0002: │ @types/copy-webpack-plugin@npm:10.1.0 doesn't provide webpack (p77c89), requested by copy-webpack-plugin
➤ YN0002: │ benchmark@workspace:benchmark doesn't provide react-refresh (pcbcae), requested by @pmmmwh/react-refresh-webpack-plugin
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide @types/node (p3f67f), requested by ts-node
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react (paeba2), requested by @storybook/react
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react (p91125), requested by @storybook/addon-essentials
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react (pacf49), requested by @storybook/react-webpack5
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react-dom (pe7937), requested by @storybook/react
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react-dom (pbf2fb), requested by @storybook/addon-essentials
➤ YN0002: │ foxglove-studio@workspace:. doesn't provide react-dom (p975b8), requested by @storybook/react-webpack5
➤ YN0002: │ notistack@npm:3.0.1 [6eb1c] doesn't provide csstype (p89136), requested by goober
➤ YN0002: │ react-dnd-multi-backend@npm:6.0.2 [5783c] doesn't provide react-dnd (pef277), requested by react-dnd-preview
➤ YN0060: │ react-mosaic-component@npm:5.3.0 [6eb1c] provides react-dnd-html5-backend (pe4177) with version 14.1.0, which doesn't satisfy what react-dnd-multi-backend requests
➤ YN0060: │ react-mosaic-component@npm:5.3.0 [6eb1c] provides react-dnd-touch-backend (p6bdb4) with version 14.1.1, which doesn't satisfy what react-dnd-multi-backend requests
➤ YN0002: │ react-mosaic-component@npm:5.3.0 [6eb1c] doesn't provide react-dom (pf40d2), requested by react-dnd-multi-backend
➤ YN0000: │ Some peer dependencies are incorrectly met; run yarn explain peer-requirements <hash> for details, where <hash> is the six-letter p-prefixed code
➤ YN0000: └ Completed in 0s 824ms
➤ YN0000: ┌ Fetch step
➤ YN0000: └ Completed in 2s 39ms
➤ YN0000: ┌ Link step
➤ YN0007: │ electron@npm:24.1.0 must be built because it never has been before or the last one failed
➤ YN0007: │ esbuild@npm:0.16.17 must be built because it never has been before or the last one failed
➤ YN0007: │ protobufjs@patch:protobufjs@npm%3A7.2.3#../../patches/protobufjs.patch::version=7.2.3&hash=b342bd&locator=%40foxglove%2Fmcap-support%40workspace%3Apackages%2Fmcap-support must be built because it never has been before or the last one failed
➤ YN0007: │ playwright@npm:1.32.2 must be built because it never has been before or the last one failed
➤ YN0007: │ esbuild@npm:0.17.16 must be built because it never has been before or the last one failed
➤ YN0007: │ core-js@npm:3.19.2 must be built because it never has been before or the last one failed
➤ YN0007: │ core-js-pure@npm:3.27.1 must be built because it never has been before or the last one failed
➤ YN0007: │ lzma-native@npm:8.0.6 must be built because it never has been before or the last one failed
➤ YN0009: │ electron@npm:24.1.0 couldn't be built successfully (exit code 1, logs can be found here: /tmp/xfs-ad9b0444/build.log)
➤ YN0000: └ Completed in 1m 3s
➤ YN0000: Failed with errors in 1m 7s

**Expected Behavior**

jtbandes, 2023-04-25:
What is shown in `/tmp/xfs-ad9b0444/build.log`?

amacneil, 2023-04-27:
Closing due to insufficient information and no response.

@WangShiJieAOTM this is the relevant error - please expand it, and also check the log file @jtbandes mentioned. 

> YN0009: │ electron@npm:24.1.0 couldn't be built successfully (exit code 1, logs can be found here: /tmp/xfs-ad9b0444/build.log)

If you can't figure it out please add more details below so we can help.

closed 2023-04-27 by amacneil

WangShiJieAOTM, 2023-05-22:
# This file contains the result of Yarn building a package (electron@npm:24.3.1)
# Script name: postinstall

Error: Manifest not found
    at el.find (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:390:29961)
    at async /home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:422:4337
    at async On.openPromise (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:377:45853)
    at async p9 (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:422:3802)
    at async /home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:422:2580
    at async $t.mktempPromise (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:377:69349)
    at async c0 (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:422:2520)
    at async /home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:442:235
    at async $t.mktempPromise (/home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:377:69349)
    at async /home/ecs-user/.cache/node/corepack/yarn/3.3.1/yarn.js:442:25