foxglove/studio #5862: Incorrect visualization when plotting data with the `[:]` accessor

vhwanger, 2023-04-27:
**Description**
I have data of the form:

```
  1 syntax = "proto3";
  2
  3
  4 message ThreeTuple {
  5     float a = 1;
  6     float b = 2;
  7     float c = 3;
  8 }
  9 message DataCollection {
 10     repeated ThreeTuple data = 1;
 11 }
```

When I try and plot `/data.data[:].a` with the X-axis value using `Index` and I scrub around, the plot seemingly starts accumulating all the lines over the times I scrub over. The expected behavior is that it only plots the lines over the current timestamp of the log

[image]

- Version: 1.51.0
- Platform: Mac
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
mcap file

**Expected Behavior**
The expected behavior is shown below.
[image]

vhwanger, 2023-04-27:
I have a layout and an mcap available for reproducing if needed.

foxymiles, 2023-04-28:
Thanks for the detailed bug report. I'm seeing some strange behavior too but a bit different from what you describe. Does what you're seeing resemble this?

https://user-images.githubusercontent.com/93935560/235024939-8d47b596-1586-45b7-b9ea-52455ca3a1bf.mov

vhwanger, 2023-04-28:
That's actually what the data looks like. Now if you scrub back and forth in the window, does it look the same?

foxymiles, 2023-04-28:
Yes I see what you mean. Scrubbing on the timeline back and forth results in these lines accumulating like this.

foxymiles, 2023-04-28:
@vhwanger Could you please try the deployment from this PR: https://github.com/foxglove/studio/pull/5868 and see if it fixes your problem? You can try it here: https://studio-8uik5ytns.foxglove.party/

vhwanger, 2023-04-28:
This looks like it fixes it on my fake data file, thank you so much! I'll have to wait for the desktop deployment to try it on the actual log I have.

foxymiles, 2023-07-12:
Looks like I inadvertently broke this as part of recent optimization work.

PR #5868 (2023-05-01): Skip current frame accumulation for non-timebased plots
**User-Facing Changes**
Fix an issue with timeline scrubbing on non-time based plots.

**Description**
The logic we added in #5694 for accumulating currentFrame messages across multiple `useMessageReducer` invocations assumes we're plotting by time. For plots with non-time based X axes this is incorrect and we should only include the most recent frame in the data.

Fixes #5862
closed 2023-07-12 by foxymiles