foxglove/studio #5870: No coordinate frames found

ClemensLinnhoff, 2023-04-28:
**Description**
I would like to use foxglove studio to visualize Marker and PointClouds from ROS2 Humble. Unfortunately, I cannot select a Display Frame for the 3D panel, as you can see in the screenshot. However, the tf is transferred and I can select the available frames in rviz2, as displayed in the second screenshot.

[image]

[image]

- Version: Foxglove Studio 1.52.0
- Platform: Ubuntu 22.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): foxglove_bridge

**Steps To Reproduce**

**Expected Behavior**
Being able to select the Display Frame as I can in rviz2.

jtbandes, 2023-04-28:
It looks like none of the TF frames are visible in Studio at all. What do you see in the list of topics in the app, under the data source tab? Are you able to view any TF messages with Raw Messages? Does the foxglove_bridge process have any log messages in its console?

ClemensLinnhoff, 2023-04-28:
That's odd, once I opened the data source tab, two (of four) frames appeared:

[image]

Now I can display the PointCloud in the base_link frame. But markers given in world frame are missing. The error message says: "Missing transform from frame <world> to frame <base_link>". But it is part of the /tf message.

jtbandes, 2023-04-28:
How frequently is `/tf` being published?

If you record a mcap file and load that directly into Studio, can you reproduce the same problems or does it only happen with a live connection?

Any sample data you can provide or steps to reproduce would be helpful in debugging.

ClemensLinnhoff, 2023-05-03:
The /tf is published every time step, so 10 times per second.
Interestingly it works fine with the mcap. I can select all frames and they are displayed correctly. It just does not work live.
Unfortunately I cannot share data or code.

achim-k, 2023-05-03:
Do you see any warnings/errors (e.g. `Send buffer limit reached`) on the foxglove bridge side?

achim-k, 2023-05-03:
This error is unrelated. I was wondering if maybe the server fails to communicate some `/tf` messages to Studio, but if you don't see a `Send buffer limit reached` warning, then that should not be the case.

achim-k, 2023-05-05:
> This error is unrelated. I was wondering if maybe the server fails to communicate some `/tf` messages to Studio, but if you don't see a `Send buffer limit reached` warning, then that should not be the case.

I have just discovered a bug in foxglove_bridge which prevented the `Send buffer limit reached` warning from being logged. So it could be that the server was dropping messages but it was not obvious that it was doing so.
See also https://github.com/foxglove/ros-foxglove-bridge/pull/224

ClemensLinnhoff, 2023-06-07:
Yes, this is still kind of an issue. I can now select the different display frames and also view the data in the respective frames. But I cannot view all data in a single frame. It shows a warning, that the transforms are missing. Maybe I am doing something wrong with the transforms, but to my knowledge, they should be all there and be correct. Anyways, rviz2 does not have a Problem with it and can visualize all data from all topics correctly in all frames.

snosenzo, 2023-06-07:
@ClemensLinnhoff We'd really like to help you out further but unfortunately it's a bit hard to do without sample data. If you'd be comfortable joining our public slack channel and DMing me or emailing our support (support [at] foxglove.dev) mentioning this issue with some sample data, we'd be happy to look into this further.

snosenzo, 2023-06-14:
Going to close this issue for now since I haven't heard back, but anyone should feel free to re-open if they have some sample data that can help us reproduce this

snosenzo, 2023-06-20:
Closing since we haven't heard back and there's not much more we can do with this. But anyone should feel free to re-open if they're able to share some data

closed 2023-06-20 by snosenzo

achim-k, 2023-06-28:
Might be related: https://github.com/foxglove/studio/issues/6309

abrajes, 2023-08-25:
Hello everyone!

I am trying to create a mcap file from data that was stored in csv, and for that I used the code from the following link
https://github.com/foxglove/mcap/blob/main/python/examples/jsonschema/pointcloud_csv_to_mcap.py

I was able to generate a mcap file but I get the same error for display frame, when I tried to visualize the data in foxglove
[image]

Can someone help me solve this problem.
Thanks

jtbandes, 2023-08-25:
Click the 👁️ to enable your point cloud topic.

abrajes, 2023-08-29:
Thanks you @jtbandes
I hadn't noticed that, thanks for helping out. Now, I am able to visualize data given in the example code