foxglove/studio #5877: Indicator panel not compatible with array message types

antrose99, 2023-04-29:
**Description**

This is a bit of a janky setup but I am using the sensor_msgs/msg/Imu message for INS data and storing the status message (which is just a number) in one of the covariances, which is defined as a Float64[9] message.

To visualize this status message in Foxglove I wanted to use the indicator panel to display the appropriate label and color for the given status. Entering all my rules and assigning the message path to, in my case, /imu/inspva.twist.covariance[0], led to the indicator falling back to the "Otherwise" rule (even if I set a rules that encapsulates all numbers e.g. one rule for >0 and one for <= 0). Using the raw panel with the same message path shows just the number (and not an array), as expected, and I can confirm that the indicator works fine for the other messages of the Imu message.

Version: 1.52.0
Platform: MCAP
Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
mcap file

**Steps To Reproduce**

1. Set the message path to one element of an array message in the indicator panel.

2. Add rules covering the entire space, e.g. for a float >0 and <= 0.

3. Notice that that the indicator falls back to the "Otherwise" case

Can provide an mcap containing messages with this problem if needed.

**Expected Behavior**

That it would use the data from the element in the array that was specified as the value the indicator compares against.

jtbandes, 2023-04-29:
> Can provide an mcap containing messages with this problem if needed.

Yes please!

antrose99, 2023-04-30:
Here is the mcap file: [indicator_array_mcap.zip](https://github.com/foxglove/studio/files/11360318/indicator_array_mcap.zip)

And here is a screenshot which perhaps better visualizes the behavior
[image]

closed 2023-05-23 by wimagguc

pegadoflavian, 2023-05-31:
Thank you for fixing this :)