foxglove/studio #5935: 3D crashed

263868161, 2023-05-06:
**Description**

When I use 3D, a crash occurs.Works fine on older versions（1.2.0）.
[image]
[image]
[image]

- Version:1.53.0
- Platform:ubuntu20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

**Expected Behavior**
[demo.zip](https://github.com/foxglove/studio/files/11412196/demo.zip)

defunctzombie, 2023-05-07:
@263868161 What type of computer are you running this on? What are the graphics drivers you are using? To me this looks like a limitation with whatever graphics drivers you are using on your specific computer.

263868161, 2023-05-08:
My GPU is nvidia  GEFORCE GTX 1060, gpaphics drivers is "nvidia-driver-525"

263868161, 2023-05-09:
@defunctzombie Hi,I have uploaded the demo program, can't it run?

jtbandes, 2023-06-10:
Hi again – we've made some performance improvements in #6034 that should help with your use case. I'm going to close out this issue for now since we haven't heard back recently, but feel free to report back or create a new issue if you are still having problems. (A sample bag/mcap file will be much easier for us to diagnose with than a sample program.)

closed 2023-06-10 by jtbandes