foxglove/studio #6016: Undesirable behavior when robot_description topic is remapped

adamconkey, 2023-05-12:
**Description**
It seems currently the built-in support for visualizing a URDF in the 3D panel from a topic assumes that the only valid topic one can publish a URDF to (and in turn visualize) is `robot_description`. However, in ROS2 nodes support `remapping` of topics, so I can for example do the following in my launch file:

```python
robot_state_publisher_node = Node(
    name='robot_state_publisher',
    package='robot_state_publisher',
    executable='robot_state_publisher',
    remappings=[
        ('/joint_states', '/pseudo_joint_states'),
        ('/robot_description', '/custom_robot_description')
    ],
)
```

In Foxglove Studio, I then lose the ability to visualize that URDF, only because it's on a topic different from `robot_description`. This is undesirable, since if I have two robots I want to visualize at once, each with its own robot description topic and joint state topic, it's unclear how to visualize them both.

A huge improvement would be to have any topic that is having a URDF being published to it be picked up as a candidate for visualization in the 3D panel, and have it visualize URDF with the corresponding joint states (which may have also been remapped when the robot state publisher node was created).

- Version: 1.53.1
- Platform: Mac Desktop
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS2, websockets, running on remote Linux

**Steps To Reproduce**

Create a robot state publisher node like the following in a launch file:
```python
robot_state_publisher_node = Node(
    name='robot_state_publisher',
    package='robot_state_publisher',
    executable='robot_state_publisher',
    remappings=[
        ('/joint_states', '/pseudo_joint_states'),
        ('/robot_description', '/custom_robot_description')
    ],
)
```

**Expected Behavior**

In Foxglove Studio I would see an option in the `Topic` dropdown for the `3D panel settings` for the `/custom_robot_description` topic, and that it would be a candidate for visualizing my URDF in the 3D panel, visualizing joint states from the `/pseudo_joint_states` topic, as prescribed by the remappings in the node creation.

**Observed Behavior**
There is no option to visualize my URDF in the 3D panel via a topic, now that `/robot_description` was remapped to `/custom_robot_description`, and Foxglove Studio assumes it can only be `/robot_description`.

defunctzombie, 2023-05-12:
> topic that is having a URDF being published to it be picked up as a candidate

One gotcha I will note is that there is no way to identify any such topic. These URDF description topics all are a generic "string" type last time I checked into this.

jtbandes, 2023-05-12:
I'm going to remove the bug label because I don't believe this is a bug in the implementation, simply a limitation with the current design of the feature.