foxglove/studio #6033: Image marker persists even though empty value is returned by user script

vivek-rane, 2023-05-16:
**Description**
When a 2D image marker is returned as empty `ImageAnnotations` by the user script, the last seen marker continues to be drawn on the screen. This finally disappears when a populated value is returned by the script and is replaced with the updated marker.

Version: 1.53.1 (darwin)
Platform: macOS Monterey (12.5.1)

**Steps To Reproduce**
Download the test mcap from https://github.com/foxglove/mcap/blob/main/testdata/mcap/demo.mcap
Add the following user script:
```
import { Input, Message } from "./types";

export const inputs = ["/image_color/compressed"];
export const output = "test marker";

type ImageMarker = Message<"foxglove.ImageAnnotations">;

var color_val = { r: 1, g: 0, b: 0.5, a: 1 };

export default function node(
  event: Input<"/image_color/compressed">
): ImageMarker {
  if (
    1490149582 < event.receiveTime.sec &&
    event.receiveTime.sec < 1490149586 
  ) {
    return <ImageMarker>{};
  }
  return {
    circles: [
      {
        timestamp: event.receiveTime,
        thickness: 2,
        position: { x: (event.message.header.seq - 100) * 10, y: 300 },
        outline_color: color_val,
        fill_color: color_val,
        diameter: 100,
      },
    ],
    points: [],
    texts: [],
  };
}
```
Open an image panel and point it to /image_color/compressed
Turn on the marker called "test marker" from our script
Start playback

**Expected Behavior**
When the script returns an empty marker, it should disappear from the image.

https://github.com/foxglove/studio/assets/25373098/fe90f061-6481-4bbb-8d4e-be386bc475b8

vivek-rane, 2023-05-17:
On a related note, would it be possible to have a time-to-live similar to the 3D `SceneUpdate` messages?

snosenzo, 2023-06-01:
Thanks for filing! A fix has been implemented in the linked PR.

> On a related note, would it be possible to have a time-to-live similar to the 3D SceneUpdate messages?

Definitely possible! If you could open up a discussion over [here](https://github.com/orgs/foxglove/discussions) we'd be happy to consider and discuss this request further

closed 2023-06-08 by snosenzo