foxglove/studio #6062: Constant erro when connected to wifi without internet

bertaveira, 2023-05-18:
**Description**
We connect to our robots wifi where we use Foxglove Bridge. The problem is that this wifi does not have an internet connection. Because of this I get a constant error like this:
[image]

It becomes quite disturbing to have to close it every few seconds

- Version: 1.53.1
- Platform: MacOS
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): Foxglove Rosbridge

foxymiles, 2023-07-18:
Handing off to @2metres for UX inspiration here.

2metres, 2023-08-11:
I think we should persistant errors into the Problems sidebar instead of pushing to a toast

foxymiles, 2023-09-14:
These will show up as items in the problems sidebar instead of as a snackbar in the next studio release.

closed 2023-09-14 by foxymiles