foxglove/studio #6082: `foxglove.Log` messages are dropped when recording/publish time is outlier

eewanco, 2023-05-22:
**Description**
When an MCAP file has multiple `foxglove.Log` schema messages, messages with a recording/publish timestamp that is an outlier, even by a relatively small amount (e.g. five minutes), when the Log Panel is opened, only a subset of the log messages are displayed.

- Version: 1.54.0
- Platform: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): MCAP

**Steps To Reproduce**

1. Run the enclosed Python program (`python3 foxglove-bug.py`). It will generate three files: `bug_log1.mcap`, `bug_log2.mcap`, and `bug_log3.mcap`.
2. Merge the files: `mcap merge bug_log[1-3].mcap > bug_log_merged.mcap`
3. Launch a clean instance of Foxglove Studio 1.54.0
4. Load `bug_log_merged.mcap`
5. Create new layout from "Log" template
6. Observe that only a subset of log entries are shown in log panel
7. Cross-check with ` mcap info bug_log_merged.mcap ` or `mcap cat --json bug_log_merged.mcap`
8. Uncomment the other failing timestamp and compare
9. Uncomment the successful timestamp and compare

**Expected Behavior**

All three log entries should be shown.

[timestamp-bug.zip](https://github.com/foxglove/studio/files/11533747/timestamp-bug.zip)

eewanco, 2023-05-22:
This may be layout-related. I observed a slight change in behavior between the time I used my own custom layout and the time that I reloaded with a new layout; both malfunctioned, but in different ways (specifically, my custom layout showed two entries with the 1675381583628179000 timestamp, and the new layout from the Log template after relaunching shows only one). I could not reproduce the exact original behavior by reloading my old layout.

bryfox, 2023-05-26:
> messages with a recording/publish timestamp that is an outlier, even by a relatively small amount (e.g. five minutes), when the Log Panel is opened, only a subset of the log messages are displayed.

Are you playing back the file? The log panel does not display all messages up front but will append messages as playback continues. Your third message occurs near the end of that timeline, so I'd expect it to display once playback reaches that point or you seek to the end.

bryfox, 2023-05-26:
> after relaunching shows only one

I suspect what's happening in this case is that once you've loaded the first time, Studio will save your seek position (as a `time` query parameter). When you refresh, Studio begins from that seek time by backfilling a single message on the topic.

eewanco, 2023-05-26:
I see your point. This was an attempt to reduce to a minimal reproducible example a more complex problem. It looks like I inadvertently stumbled on the wrong path and will have to backtrack and regroup. Although in attempting to do so, I think I confirmed that the issue is probably the same; I have merged three files, two of which are near in time and one is far off in time. I was hoping by stamping the logs with a 0 (1-jan-1970) timestamp they would be shown first (this is not shown in the sample data I furnished here), but I see the actual algorithm makes a certain amount of sense. (What is the `Log.timestamp` field used for, other than display, anything?) I think that combining the fact that I am attempting to mix, for test purposes, data that would not be combined in the real world, and the fact that I have probably misused the feature (I basically wanted to write a comment into the file), makes me conclude that I should probably abandon my hopes of getting what I want accomplished.

bryfox, 2023-05-26:
I see. Yes, I believe that timestamp is only for display.

closed 2023-05-30 by jtbandes