foxglove/studio #6144: NumberInput component reallyValue no same as DisplayValue

smbera, 2023-05-30:
**Description**

- Version: 1.54.1-dev
- Platform: Google Chrome(112.0.5615.165)
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

Here is the PR https://github.com/foxglove/studio/pull/6143

**Steps To Reproduce**

[image]

**Expected Behavior**
[image]

foxymiles, 2023-06-05:
This is not a bug. By default the number input in settings shows a precision of 2:

https://github.com/foxglove/studio/blob/54a1534574d97a5bcb60083d7065b507af0daa1d/packages/studio-base/src/components/SettingsTreeEditor/inputs/NumberInput.tsx#L84

closed 2023-06-05 by foxymiles

smbera, 2023-06-05:
> This is not a bug. By default the number input in settings shows a precision of 2:
> 
> https://github.com/foxglove/studio/blob/54a1534574d97a5bcb60083d7065b507af0daa1d/packages/studio-base/src/components/SettingsTreeEditor/inputs/NumberInput.tsx#L84

Thanks for you apply. I know precision is 2 default. 
Actually, but the precision is not 2 decimal places. You can see my PR  https://github.com/foxglove/studio/pull/6143