foxglove/studio #6169: Char arrays not displaying properly

viveksood97, 2023-06-01:
**Description**
Char arrays are being processed and displayed as uint8 arrays Check the name field
[image]

Echoing the topic in CLI displays the same field correctly. Check the name field
[image]

- Version:
- Platform:
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):

**Steps To Reproduce**

**Expected Behavior**
Display char arrays as char arrays

defunctzombie, 2023-06-01:
Please clarify what you mean by "char arrays"? Which serialization format are you using? What does your message schema/definition look like?

jtbandes, 2023-06-01:
Please remember to fill out all fields of the issue template when filing a new issue.

viveksood97, 2023-06-01:
@defunctzombie 
1. By char arrays I mean: 
```
# Task (abbreviated) name, including null termination character
char[6] name
```
2. I am not sure about the serialization format. If you could shed a little bit more light on how to figure that out, I'd be happy to provide that information.
3. Here's the output of rosmsg show:
[image]

jtbandes, 2023-06-01:
According to http://wiki.ros.org/msg, `char` is a "deprecated alias for uint8". This is why the app displays `char[]` as a Uint8Array. Generally, arrays of uint8 values may not form a valid string. For strings, use the `string` primitive type.

viveksood97, 2023-06-01:
Closing the issue, thanks!!

closed 2023-06-01 by viveksood97