foxglove/studio #6224: Error: invalid wire type 7 at offset 192

damonChenzf07, 2023-06-12:
**Description**

- Version: 1.56.0
- Platform: Ubuntu
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
ws-websocket server(c++)

**Steps To Reproduce**

we have a image message type:(use protobuf version 2)
message CompressedImage {
  optional apollo.common.Header header = 1;
  optional string frame_id = 2;

  // Specifies the format of the data
  //  Acceptable values: jpeg, png
  optional string format = 3;
  optional bytes data = 4;  // Compressed image buffer
  optional double measurement_time = 5;
  optional uint32 frame_type = 6;
}

than, we send this message buffer to studio client, the client got an error:
Error: **invalid wire type 7 at offset 192**
    at M.skipType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1228:55731)
    at M.skipType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1228:55673)
    at M.skipType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1228:55673)
    at M.skipType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1228:55673)
    at C.Header$decode [as decode] (eval at O (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:388:23427), <anonymous>:47:5)
    at C.CompressedImage$decode [as decode] (eval at O (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:388:23427), <anonymous>:11:21)
    at Object.P [as deserialize] (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4829:258599)
    at r.<anonymous> (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4830:4383)
    at r.emit (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:1133:7318)
    at ws.onmessage (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4829:264786)
    at IB.#e.onmessage (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/main.js:4829:275976)

but we send another topic data, and its message type as follow:(TransformStampeds)
message Transform {
  optional apollo.common.Point3D translation = 1;
  optional apollo.common.Quaternion rotation = 2;
}

message TransformStamped {
  optional apollo.common.Header header = 1;
  optional string child_frame_id = 2;
  optional Transform transform = 3;
}

message TransformStampeds {
  optional apollo.common.Header header = 1;
  repeated TransformStamped transforms = 2;
}

the studio can correct parse this message.

the apollo.common.Header type defines here:

message Header {
  // Message publishing time in seconds.
  optional double timestamp_sec = 1;

  // Module name.
  optional string module_name = 2;

  // Sequence number for each message. Each module maintains its own counter for
  // sequence_num, always starting from 1 on boot.
  optional uint32 sequence_num = 3;

  // Lidar Sensor timestamp for nano-second.
  optional uint64 lidar_timestamp = 4;

  // Camera Sensor timestamp for nano-second.
  optional uint64 camera_timestamp = 5;

  // Radar Sensor timestamp for nano-second.
  optional uint64 radar_timestamp = 6;

  // data version
  optional uint32 version = 7 [default = 1];

  optional StatusPb status = 8;

  optional string frame_id = 9;
}

all the ws-websocket server side protocobuf use the version 2.

**Expected Behavior**

the Image message can also work.

jtbandes, 2023-06-12:
Are you able to produce a .mcap file that contains this data so we can reproduce the issue locally without trying to guess how you've set up your server?

You can record a mcap file from your existing server using:
`npx @foxglove/ws-protocol-examples@latest mcap-record ws://localhost:8765 -o <file.mcap>`

achim-k, 2023-06-22:
@damonChenzf07 is this still an issue for you? If so, could you share an MCAP file with some sample data (see comment above)?

jtbandes, 2023-06-27:
Since we haven't heard back for a couple of weeks I'm going to close this out. Feel free to update the issue if you're able to share some sample data, and we'd be happy to help take a closer look.

closed 2023-06-27 by jtbandes