foxglove/studio #6250: Playback .db3 recorded fellowing foxglove ros2 message schema

XunHang, 2023-06-14:
**Description**

when I open a .db3 file which contain topic 'foxglove/msg/SceneUpdate' by foxglove studio, the studio gives me an error below:

> Error: Unrecognized primitive type builtin_interfaces/Time
    at A.readComplexType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/6938.js:1:13101)
    at A.readComplexType (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/6938.js:1:12801)
    at A.readMessage (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/6938.js:1:12470)
    at m.decodeMessage [as decoder] (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/4471.js:5:3406)
    at m.next (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/4471.js:5:1501)
    at async R.messageIterator (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/4824.js:1:3657)
    at async h.next (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/4824.js:1:513)
    at async h.nextBatch (file:///opt/Foxglove%20Studio/resources/app.asar/renderer/4824.js:1:568)

- Version: 1.54.0
- Platform: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
[test_SceneUpdate.db3](https://drive.google.com/file/d/1NT0-fUqgHWUVpezzXAbrYxixMFe0unhY/view?usp=sharing)

**Steps To Reproduce**

I think the visual schema design of foxglove is very good, so I record my ros2 data directly according to the message definition of foxglove. My recording code is as follows:

```c++
// create a message by foxglove protobuf schama
auto test_proto = new foxglove::SceneUpdate();
// add an empty entity, just set id & frame id
auto test_entity = test_proto->add_entities();
test_entity->set_frame_id("root");
test_entity->set_id("test");

// some operation to convert protobuf to ros2 cdr data, then record it
// ...
// timestamp is 'now'
db3_bag_->Write(test_proto, "/test/SceneUpdate",
                      "foxglove_msgs/msg/SceneUpdate", timestamp);
```

Then, I use ros2 to view my data, and I can parse out the content normally;
```shell
>ros2 bag info test_SceneUpdate.db3       
[INFO] [1686731176.931226879] [rosbag2_storage]: Opened database 'test_SceneUpdate.db3' for READ_ONLY.

Files:             test_SceneUpdate.db3
Bag size:          36.0 KiB
Storage id:        sqlite3
Duration:          27.111s
Start:             Apr 26 2023 17:22:09.86 (1682500929.86)
End:               Apr 26 2023 17:22:36.198 (1682500956.198)
Messages:          272
Topic information: Topic: /test/SceneUpdate | Type: foxglove_msgs/msg/SceneUpdate | Count: 272 | Serialization Format: cdr
```
But when I use foxglove studio load this .db3 file, I get this problem.

I have done this with foxglove.Vector2 before, just use `db3_bag_->Write(test_proto, "/test/Vector2",
                      "foxglove_msgs/msg/Vector2", timestamp);` and the studio can correctly parse and draw the plot.

So I looked at the specific schema definition and found something that seems to be related. In file [ros2/SceneEntity.msg](https://github.com/foxglove/schemas/blob/main/ros_foxglove_msgs/ros2/SceneEntity.msg), **timestamp**'s type is '**builtin_interfaces/Time**'.

I don't know if this is the right direction, hoping to help the foxglove team locate the problem.

**Expected Behavior**
Locate the problem, I don't have any front-end capabilities, and I don't know where to start when facing the studio

jtbandes, 2023-06-14:
Hi,
Storing Protobuf messages in a .db3 bag with `Serialization Format: cdr` is not supported. I would recommend using a MCAP file instead: https://mcap.dev/guides/cpp/protobuf#writing

XunHang, 2023-06-15:
1. `// some operation to convert protobuf to ros2 cdr data` is a code section I wrote before, and it works here, so I copy it to my new code.
2. I have shared my db3 file in **Description**, the 
   - Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
[test_SceneUpdate.db3](https://drive.google.com/file/d/1NT0-fUqgHWUVpezzXAbrYxixMFe0unhY/view?usp=sharing)

    I upload it to google drive and set it can be download.
3. I know mcap is recommended, but I still want to try .db3 cause my db3 record code is work. (*^▽^*)

XunHang, 2023-06-15:
Sorry, I didn't see some question you asked.
1. I don't install `ros-$ROS_DISTRO-foxglove-msgs` from apt;
2. I use ros2-galactic to develop;

In addition, in order to prove that my db3 recording code is correct, I recorded the data of foxglove::Vector2, which can be correctly recognized and plotted by foxglove studio, and I will also share it with you.

*db3 file*
[test_Vector2.db3](https://drive.google.com/file/d/1AUMfdqzvzq3z0FjqxEt_yz4DyaI00KR7/view?usp=sharing)

*cpp cpde*
``` cpp
// create a message by foxglove protobuf schama
      auto test_proto = new foxglove::Vector2();
      test_proto->set_x(1.0);
      test_proto->set_y(1.5);

// some operation to convert protobuf to ros2 cdr data, then record it
// ...
// timestamp is 'now'
db3_bag_->Write(visual_proto, "/test/Vector2",
                      "foxglove_msgs/msg/Vector2", timestamp);
```
*ros2 bag info*
```shell
[INFO] [1686790366.244169571] [rosbag2_storage]: Opened database 'test_Vector2.db3' for READ_ONLY.

Files:             test_Vector2.db3
Bag size:          36.0 KiB
Storage id:        sqlite3
Duration:          27.111s
Start:             Apr 26 2023 17:22:09.86 (1682500929.86)
End:               Apr 26 2023 17:22:36.198 (1682500956.198)
Messages:          272
Topic information: Topic: /test/Vector2 | Type: foxglove_msgs/msg/Vector2 | Count: 272 | Serialization Format: cdr
```

PR #6343 (2023-06-30): Bump @foxglove/rosbag2-web from 4.1.0 to 4.1.1
**User-Facing Changes**
Fix invalid message definitions when loading a .db3 file containing [foxglove_msgs](https://github.com/foxglove/schemas)

**Description**
Bumps `@foxglove/rosbag2-web` which transitively includes https://github.com/foxglove/schemas/pull/115. This fixes #6250.
commit 55faf4b6 (2023-06-30): Bump @foxglove/rosbag2-web from 4.1.0 to 4.1.1 (#6343)

**User-Facing Changes**
Fix invalid message definitions when loading a .db3 file containing
[foxglove_msgs](https://github.com/foxglove/schemas)

**Description**
Bumps `@foxglove/rosbag2-web` which transitively includes
https://github.com/foxglove/schemas/pull/115. This fixes #6250.
closed 2023-06-30 by achim-k