foxglove/studio #6264: The plot render two frame data once

zhangfei56, 2023-06-16:
The X Axis I choose Index.
The Message path I set one array data: speedPoint[:].a

Then I Seek backward and forward more then once, the  plot may be render two frame data once.

[image]

foxymiles, 2023-06-28:
Is it possible for you to share a sample data file that demonstrates the issue?

foxymiles, 2023-07-13:
So far I haven't been able to reproduce this with test data here. Could you please try this again on studio.foxglove.dev and see if the problem persists? If so could you please also provide some test data to reproduce the issue?

foxymiles, 2023-07-18:
I haven't been able to reproduce this so I'm going to close it. I'm happy to reopen it if you can provide more detailed reproduction steps.

closed 2023-07-18 by defunctzombie