foxglove/studio #6275: Unable to Select Colour Using Keyboard

camchambers, 2023-06-21:
**Description**
Typing a hex colour in the colour picker and pressing enter does not select the colour. 

[image]

- Version:  1.54.0
- Platform: Ubuntu Linux
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): rosbridge

**Steps To Reproduce**
- Configure a topic containing a color picker
- Enter a hex value into the colour field
- Press enter

**Expected Behavior**
- The colour entered is updated in the UI

jtbandes, 2023-06-22:
Hi, I believe this is happening because the color picker requires 8 digits, `#RRGGBBAA`. However, I agree that it should be possible to enter 6 values and automatically set the alpha to FF.

camchambers, 2023-06-22:
Right. The field requires an alpha value (suggested by the placeholder value). Defaulting the alpha would be a nice feature.

foxymiles, 2023-07-03:
I'm going to reclassify this as an enhancement request.

camchambers, 2023-07-03:
Absolutely.

PR #6575 (2023-08-03): Allow more flexible color entry for ColorPicker settings item
**User-Facing Changes**
Allow more color string input variants for ColorPicker panel setting input type.

**Description**
Prior to this change, a user had to entry all of the hex digits into the color picker text input before the picker would accept their color. For RGB colors this meant entering all 6 hex characters for `rrggbb` and, for RGBA, 8 hex characters `rrggbbaa`.  When using RGBA it is common to want no opacity and default the last two characters to `ff`.

This change updates the hex text input to accept any of the followi [... cut]
commit 3f25dfdf (2023-08-03): Allow more flexible color entry for ColorPicker settings item (#6575)

Prior to this change, a user had to entry all of the hex digits into the
color picker text input before the picker would accept their color. For
RGB colors this meant entering all 6 hex characters for `rrggbb` and,
for RGBA, 8 hex characters `rrggbbaa`. When using RGBA it is common to
want no opacity and default the last two characters to `ff`.

This change updates the hex text input to accept any of the following
for RGB inputs: `rrggbb` and the common shorthand `rgb` which expands to
repeat the characters -> `rrggbb`.

An [... cut]
closed 2023-08-03 by defunctzombie