foxglove/studio #6337: Map panel errors when covariance is NaN

rgov, 2023-06-29:
We are using the [`gpsd_client`](https://github.com/swri-robotics/gps_umd/tree/master/gpsd_client) node to publish our fix from GPSD. Because our GPS receiver does not report the necessary NMEA strings to determine the fix covariance, `gpsd_client` would by default not publish any fix, so we are running the node with the `check_fix_by_variance=false` parameter.

This results in the fix message containing `position_covariance_type=COVARIANCE_TYPE_DIAGONAL_KNOWN` but `position_covariance=[NaN, 0, 0, 0, NaN, 0, 0, 0, NaN]`.

Consequently, in Studio's `getAccuracy()`, the radius is computed as `Math.sqrt(NaN)` which causes the panel to error.

https://github.com/foxglove/studio/blob/a845e11df62ffc4c92aa8160b9e7e125a99fccf3/packages/studio-base/src/panels/Map/getAccuracy.ts#L34-L39

I suppose this may be considered a bug in `gpsd_client` so I will also report it there, but the panel could handle this case.

[image]

defunctzombie, 2023-06-29:
@rgov Do you think the panel should tell you that the covariance has invalid values? silently ignore the covariance? treat NaN like 0?

rgov, 2023-06-29:
In our case I'm fine with ignoring / treating it like 0. If you displayed a small ⚠️ with a tooltip on mouseover, that's fine, but I wouldn't want it really flashy.

jtbandes, 2023-06-29:
What does "causes the panel to error" mean? I don't see a screenshot of what error you are experiencing?

rgov, 2023-06-29:
I reported this second-hand from the end user, who said "I’m getting map to flash to panel then an error ‘Circle radius cannot be NaN’.".

defunctzombie, 2023-07-11:
This was fixed in https://github.com/foxglove/studio/pull/6347

closed 2023-07-11 by defunctzombie