foxglove/studio #6348: PointCloud Flatbuffer message from Foxglove websocket cannot be visualized anymore in Foxglove Studio (v1.58+)

lhuang-ae, 2023-07-03:
PointCloud Flatbuffer message from Foxglove websocket cannot be visualized anymore in Foxglove Studio (v1.58+)

- Version: 1.58+
- Platform: (Mac + Chrome web)
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native):
```
import asyncio
import base64
import importlib.resources as pkg_resources
import random
import time

import flatbuffers
import foxglove_schemas_flatbuffer
import foxglove_schemas_flatbuffer.Time as FbTime
from foxglove_schemas_flatbuffer import (
    PackedElementField,
    PointCloud,
)

from foxglove_websocket import run_cancellable
from foxglove_websocket.server import FoxgloveServer, FoxgloveServerListener

def _float32_field(builder, name, offset):
    PackedElementField.Start(builder)
    PackedElementField.AddName(builder, name)
    PackedElementField.AddOffset(builder, offset)
    PackedElementField.AddType(builder, 7)
    return PackedElementField.End(builder)

def _build_points():
    nb_points = 10000
    builder = flatbuffers.Builder(nb_points * 12)
    x = builder.CreateString("x")
    y = builder.CreateString("y")
    z = builder.CreateString("z")

    x_buff = _float32_field(builder, x, 0)
    y_buff = _float32_field(builder, y, 4)
    z_buff = _float32_field(builder, z, 8)
    # Prepend all field descriptions into a Vector buffer in reverse order.
    pc_fields = 0
    PointCloud.StartFieldsVector(builder, 3)
    builder.PrependUOffsetTRelative(z_buff)
    builder.PrependUOffsetTRelative(y_buff)
    builder.PrependUOffsetTRelative(x_buff)
    pc_fields = builder.EndVector()

    PointCloud.StartDataVector(builder, nb_points * 12)
    for _ in range(nb_points):
        builder.PrependFloat32(random.random())
        builder.PrependFloat32(random.random())
        builder.PrependFloat32(random.random())
    pc_data = builder.EndVector()

    frame_id = builder.CreateString("root")
    PointCloud.Start(builder)
    timestamp = FbTime.CreateTime(builder, int(time.time()), 0)
    PointCloud.AddTimestamp(builder, timestamp)
    PointCloud.AddFrameId(builder, frame_id)
    PointCloud.AddPointStride(builder, 12)
    PointCloud.AddFields(builder, pc_fields)
    PointCloud.AddData(builder, pc_data)
    data_offset = PointCloud.End(builder)

    builder.Finish(data_offset)
    return builder.Output()

async def main():
    async with FoxgloveServer("0.0.0.0", 8765, "example server") as server:
        with (pkg_resources.files(foxglove_schemas_flatbuffer) / "PointCloud.bfbs").open("rb") as schema_file:
            schema_data = schema_file.read()
        pc_chan_id = await server.add_channel(
            {
                "topic": "pointcloud",
                "encoding": "flatbuffer",
                "schemaName": "foxglove.PointCloud",
                "schemaEncoding": "flatbuffer",
                "schema": base64.b64encode(schema_data).decode("ascii"),
            }
        )

        class Listener(FoxgloveServerListener):
            def on_subscribe(self, server, channel_id):
                now = time.time_ns()
                if channel_id == pc_chan_id:
                    message = _build_points()
                    _ = asyncio.create_task(server.send_message(pc_chan_id, now, message))

            def on_unsubscribe(self, server, channel_id):
                pass

        server.set_listener(Listener())
        while True:
            await asyncio.sleep(0.02)

if __name__ == "__main__":
    run_cancellable(main())
```

**Steps To Reproduce**

1. Run the server
2. Connecting the Studio App to the server.
3. Instead a point cloud, the error occurs.  
`Cannot mix BigInt and other types, use explicit conversions`

**Expected Behavior**
[image]

jtbandes, 2023-07-06:
Hi, thanks for your report. I am able to reproduce the issue. I have found a workaround you can use for now: `builder.ForceDefaults(True)`

jtbandes, 2023-07-06:
This PR should fix the issue: https://github.com/jkuszmaul/flatbuffers_reflection/pull/71

lhuang-ae, 2023-07-07:
Thanks!

> This issue was probably triggered by #6256.
> 
> This PR should fix the issue: [jkuszmaul/flatbuffers_reflection#71](https://github.com/jkuszmaul/flatbuffers_reflection/pull/71)

PR #6415 (2023-07-10): Bump flatbuffers_reflection from 0.0.6 to 0.0.7
Fixes #6348
Fixes FG-4136

Bumps [flatbuffers_reflection](https://github.com/jkuszmaul/flatbuffers_reflection) from 0.0.6 to 0.0.7.
<details>
<summary>Commits</summary>
<ul>
<li>See full diff in <a href="https://github.com/jkuszmaul/flatbuffers_reflection/commits">compare view</a></li>
</ul>
</details>
<br />

[image]

Dependabot will resolve any conflicts with this PR as long as you don't alter it yourself. You can also trigger a rebase manually by commenting `@dependabot rebase`.

[//]: # (dependabot-automerge-start)
[//]: # (dependabot-automer [... cut]
commit 02be651b (2023-07-10): Bump flatbuffers_reflection from 0.0.6 to 0.0.7 (#6415)

Fixes #6348
Fixes FG-4136

Bumps
[flatbuffers_reflection](https://github.com/jkuszmaul/flatbuffers_reflection)
from 0.0.6 to 0.0.7.
<details>
<summary>Commits</summary>
<ul>
<li>See full diff in <a
href="https://github.com/jkuszmaul/flatbuffers_reflection/commits">compare
view</a></li>
</ul>
</details>
<br />

[![Dependabot compatibility
score](https://dependabot-badges.githubapp.com/badges/compatibility_score?dependency-name=flatbuffers_reflection&package-manager=npm_and_yarn&previous-version=0.0.6&new-version=0.0.7)](https://docs.githu [... cut]