foxglove/studio #6471: Log Panel:

tonynajjar, 2023-07-19:
**Description**

- Version: 1.61.1
- Platform: Foxglove Studio, Ubuntu
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap file

**Steps To Reproduce**

Play an mcap file and let some logs in the Log panel flow in. Pause, click on a future time point, and play. You'll notice that all the previous logs disappear and only new logs show in the Log panel

**Expected Behavior**

Similarly to almost every panel (e.g. plot panel), I would expect the Log panel to show me all data **up to a certain time point** that I click on.

defunctzombie, 2023-07-25:
@tonynajjar Thanks for the feedback. We agree that having improved log display (across time) and search functionality would be great however the behavior you are experiencing in the current log panel is by design. The panel features do not currently preload all the log data like the plot panel does and instead only shows data during playback and clears this data when you seek.

I'm gonna close this as bug report as wontfix because we do not see it as a bug given the current design. We have noted the feature requested and if you feel inclined we'd review PRs implementing these improvements but have no timeline to offer from our side on if or when we would implement this.

closed 2023-07-25 by defunctzombie

tonynajjar, 2023-07-26:
Hi @defunctzombie, thanks for your answer. 

> we have noted the feature requested

Is this anywhere public that I can see? If not, maybe this ticket can just be rebranded as a feature request instead of closing? That way others with the same wish could upvote it

defunctzombie, 2023-07-26:
@tonynajjar There's no public link to our internal tracking issues. If you want to rally community feedback I'd suggest opening a discussion here: https://github.com/orgs/foxglove/discussions and we can link our internal tracking issue to that.