foxglove/studio #6519: Mesh Error

cody-sheff, 2023-07-26:
**Description**

- Version: 1.62+
- Platform: Ubuntu 20
- Data source: MCAP file/websocket

**Steps To Reproduce**
We've been sending geometry messages using ROS Markers that have a path to a local mesh. As of 1.62, Foxglove is consistently registering those filepaths as non-existant. To reproduce, send ROS marker messages with the mesh filename as a path to a local file. You may need to send several different meshes; it seems to *sometimes* work, but not consistently.

**Expected Behavior**
ROS Marker messages with a local filepath will be usable in Foxglove. We've verified that 1.62+ is what breaks this; rolling back to 1.61.1 restores functionality.

achim-k, 2023-07-26:
>  To reproduce, send ROS marker messages with the mesh filename as a path to a local file.

What do you mean with a path to a local file? `file://` or `package://` ?

Are you using Studio Desktop? Could share a MCAP file to reproduce this issue?

achim-k, 2023-07-26:
> Just a standard filepath, `/path/to/local/mesh.dae` style. Is the `file://` prefix needed now?

Pretty sure it works with the `file://` prefix, but let me check. If it worked before without it, then that's clearly a regression

cody-sheff, 2023-07-26:
Looks like adding the `file://` prefix does work!

achim-k, 2023-07-26:
> Looks like adding the `file://` prefix does work!

I recommend to use that from now on. I have submitted #6522 to restore the original behavior, but using a valid url (e.g. `file://`) is definitely more future proof.

achim-k, 2023-07-27:
> > Looks like adding the `file://` prefix does work!
> 
> I recommend to use that from now on. I have submitted #6522 to restore the original behavior, but using a valid url (e.g. `file://`) is definitely more future proof.

Update: We agreed to not support invalid URIs, so you will have to add the `file://` prefix from now on. Mesh resources have to be referenced by a valid URL, see also https://wiki.ros.org/rviz/DisplayTypes/Marker#Mesh_Resource_.28MESH_RESOURCE.3D10.29_.5B1.1.2B-.5D:

> The format is the URI-form used by [resource_retriever](https://wiki.ros.org/resource_retriever), including the package:// syntax.

PR #6522 (2023-07-27): Add check for fetch response status when fetching assets
**User-Facing Changes**
Not worth being mentioned in release notes

**Description**
~We were rejecting any asset uri that was not a valid URL. However, the Desktop app is able to resolve assets given as file path (e.g. `/home/foo/mesh.dae`). This patch changes the behavior to only throw an invalid URL error if it's not a desktop app.~

~Fixes #6519~

Update: We agreed on leaving it as it was and not support invalid URIs. This PR now only adds an additional check for the fetch response status.
commit 9e1037a0 (2023-07-27): Add check for fetch response status when fetching assets (#6522)

**User-Facing Changes**
Not worth being mentioned in release notes

**Description**
~We were rejecting any asset uri that was not a valid URL. However, the
Desktop app is able to resolve assets given as file path (e.g.
`/home/foo/mesh.dae`). This patch changes the behavior to only throw an
invalid URL error if it's not a desktop app.~

~Fixes #6519~

Update: We agreed on leaving it as it was and not support invalid URIs.
This PR now only adds an additional check for the fetch response status.
closed 2023-07-27 by achim-k