foxglove/studio #6532: Studio UI get stuck with a ros1 connection after Studio window is hidden for long-time time.

qdhuxp, 2023-07-27:
**Description**
Make Studio connect to a running ROS 1 system with a rosbag playing. Then minimize Studio or make studio behind other windows.
After about 5 to 10 minutes(depends on the rosbag topics data size),  show Studio, Studio UI may stuck and cannot use anymore.

- Version: v1.63.0
- Platform: Windows, Ubuntu 20
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): ROS 1 native

**Steps To Reproduce**

1. Make Studio connect to a running ROS 1 system with a rosbag playing. 
2. Then minimize Studio or make studio behind other windows.
3. After about 5 to 10 minutes(depends on the rosbag topics data size),  show Studio, Studio UI may stuck and cannot use anymore.

**Expected Behavior**
Studio should always work.

achim-k, 2023-08-09:
> After about 5 to 10 minutes(depends on the rosbag topics data size), show Studio, Studio UI may stuck and cannot use anymore.

What topic data size (bandwidth) are we talking here? Pointclouds? Would you be able to share a rosbag (and Studio layout) that we could use to reproduce the issue?

achim-k, 2023-09-11:
@qdhuxp I'm going to close this issue, feel free to reopen if this is still an issue for you

closed 2023-09-11 by achim-k