foxglove/studio #6579: Unable to use FQDN to access ROS running on remote machine

willcbaker, 2023-08-02:
**Description**

Unable to use FQDN to access ROS running on remote machine

I've noticed recently (over the last few months - this used to work, and unfortunately I don't have a version number) that I have issues using foxglove when ROS_MASTER_URI and ROS_HOSTNAME is set on the machine that requires a DNS lookup.
e.g. if ROS_MASTER_URI is set to something with an IP address everything works fine, but if I use a FQDN I get problems connecting to topics.
note: in my specific case the FQDN is a publicly available lookup pointing to an address that is inside a VPN (not publicly accessible) however this should hold for any local network as well.

- Version: 1.63.0
- Platform: Ubuntu 20.04
- Data source: ROS 1 native AND foxglove bridge

**Steps To Reproduce**

- create a resolvable hostname for a machine using a DNS server in a local network
  -  e.g. `my-robot.my-domain.org` -> `192.168.1.10`
  - `ping my-robot.my-domain.org` -> OK
- ensure that your PC is also in this network (e.g. `192.168.1.20`)

for foxglove/rosbridge:
- start the ROS + foxglove bridge on the `my-robot`
- point foxglove to ws://192.168.1.10:8765 (works)
- point foxglove to ws://my-robot.my-domain.org:8765 (doesnt work)

for ROS1 native:
- start ROS on a remote machine using `ROS_HOSTNAME=my-robot.my-domain.org` and `ROS_MASTER_URI=http://my-robot.my-domain.org:11311`
- point foxglove to ROS 1 connection at `ROS_MASTER_URI=http://my-robot.my-domain.org:11311` (doesnt work)
- point foxglove to ROS 1 connection at `ROS_MASTER_URI=http://192.168.1.10:11311` (doesnt work)

**Expected Behavior**

I expect to be able to connect and visualize the topics, instead I get errors:

```
Ensure that roscore is running and accessible at: http://skidder01.fe0414f5.polymathrobotics.dev:11311
XML-RPC call "getPublishedTopics" to http://my-robot.my-domain.org:11311 failed to connect
```

**Additional details**
Using the local addresses from example above:
I have verified the roscore is running on the REMOTE machine and able to subscribe to topics:
```
 # note this is your test machine, not where the core is running
export ROS_MASTER_URI=http://my-robot.my-domain.org:11311
export ROS_IP=192.168.1.20
rosnode list
```

achim-k, 2023-08-09:
> for foxglove/rosbridge:
> 
> * start the ROS + foxglove bridge on the `my-robot`
> * point foxglove to ws://192.168.1.10:8765 (works)
> * point foxglove to ws://my-robot.my-domain.org:8765 (doesnt work)

This should work. Can you confirm that this behavior is the same either using [websocat](https://github.com/vi/websocat) or with a regular browser? Also, I assume that the command `host my-robot.my-domain.org` correctly resolves to `192.168.1.10`?

**websocat**:
```
websocat ws://192.168.1.10:8765 --protocol foxglove.websocket.v1
websocat ws://my-robot.my-domain.org:8765 --protocol foxglove.websocket.v1
```

**Browser**: 
Copy the following code in the javascript console of an empty tab and execute it

```js
const socket = new WebSocket("ws://192.168.1.10:8765", ["foxglove.websocket.v1"]);
socket.addEventListener("open", (event) => {
  console.log("Connection to ws://192.168.1.10:8765 established")
});
const socket2 = new WebSocket("ws://my-robot.my-domain.org:8765 ", ["foxglove.websocket.v1"]);
socket2.addEventListener("open", (event) => {
  console.log("Connection to ws://my-robot.my-domain.org:8765 established")
});
```

> for ROS1 native:
> 
> * start ROS on a remote machine using `ROS_HOSTNAME=my-robot.my-domain.org` and `ROS_MASTER_URI=http://my-robot.my-domain.org:11311`

This looks incorrect. You are setting the `ROS_HOSTNAME` to the robot, but it should be the hostname of the system foxglove is running on. See also http://wiki.ros.org/ROS/EnvironmentVariables#ROS_IP.2FROS_HOSTNAME

> **Additional details** Using the local addresses from example above: I have verified the roscore is running on the REMOTE machine and able to subscribe to topics:
>
> ```
>  # note this is your test [...]

What happens if you do not set `ROS_IP`? In foxglove, you can't set the ROS_IP, hence you have to make sure that your robot can resolve the hostname of the PC you are using to connect. That's what `ROS_HOSTNAME` is for for instance. Probably related to that as well.

willcbaker, 2023-08-14:
> This looks incorrect. You are setting the ROS_HOSTNAME to the robot, but it should be the hostname of the system foxglove is running on. See also http://wiki.ros.org/ROS/EnvironmentVariables#ROS_IP.2FROS_HOSTNAME

No, the original statement is correct. This is on the _remote_ system (the robot, where the roscore is running) not the local machine (where foxglove is running)

> What happens if you do not set ROS_IP?
Then you can only subscribe to messages from robot->local, you cannot publish local->robot because the robot cannot resolve the node address.

This used to work bidirectional with foxglove where I would set ROS_HOSTNAME in the foxglove connection settings to the ROS_IP value.

## Debug

(domains and IPs redacted via find/replace)

> Also, I assume that the command host my-robot.my-domain.org correctly resolves to 192.168.1.10?

```
$host my-robot.my-domain.org
my-robot.my-domain.org has address 192.168.1.10
```

### websocat

```
$ ./websocat.aarch64-unknown-linux-musl ws://my-robot.my-domain.org:8765 --protocol foxglove.websocket.v1
```

Response (truncated):
```
{"capabilities":["clientPublish","parameters","parametersSubscribe","services","connectionGraph","assets"],"metadata":{"ROS_DISTRO":"humble"},"name":"foxglove_bridge","op":"serverInfo","sessionId":"1692041441","supportedEncodings":["cdr"]}
[WARN  websocat::readdebt] Incoming message too long (75818 > 65535): splitting it to parts.
```

### browser
```
const socket = new WebSocket("ws://192.168.1.10:8765", ["foxglove.websocket.v1"]);
socket.addEventListener("open", (event) => {
  console.log("Connection to ws://192.168.1.10:8765 established")
});
undefined
VM74:3 Connection to ws://192.168.1.10:8765 established

const socket2 = new WebSocket("ws:///my-robot.my-domain.org:8765 ", ["foxglove.websocket.v1"]);
socket2.addEventListener("open", (event) => {
  console.log("Connection to ws://my-robot.my-domain.org:8765 established")
});
undefined
VM78:1 WebSocket connection to 'ws:///my-robot.my-domain.org:8765/' failed: 
```

## Next steps

I will try this now with an additional FQDN for the local machine

achim-k, 2023-08-15:
> ### browser
> ```
> const socket = new WebSocket("ws://192.168.1.10:8765", ["foxglove.websocket.v1"]);
> socket.addEventListener("open", (event) => {
>   console.log("Connection to ws://192.168.1.10:8765 [...]

It seems that your browser already has problems in resolving `my-robot.my-domain.org`. Maybe https://bugdrivendevelopment.net/browser-ignore-internal-dns/ is your problem?

willcbaker, 2023-08-15:
I think this may be an https related issue. We are using a .dev domain which forces https.

I should have setup my example to reference a `.dev` address.

achim-k, 2023-08-17:
Closing this issue as the underlying problem does not seem related to Studio

closed 2023-08-17 by achim-k