foxglove/studio #6605: Memory leak in Studio when sending "invalid" data.

KerstinKeller, 2023-08-08:
**Description**
We have incorrectly filled PointCloud Data, and this gives a error tip, when in the 3D panel.
[image]
So far, so good.

However, as the data is not being displayed in the 3D panel, it seems to aggregate memory, and not release it, and eventually Studio crashes.
[image]

- Version: 1.63.0
- Platform: Windows
- Websocket Connection

**Steps To Reproduce**

[2023-08-08_10-45-04.461_measurement.mcap.zip](https://github.com/foxglove/studio/files/12289496/2023-08-08_10-45-04.461_measurement.mcap.zip)

Unzip the following measurement and replay with
```
npx @foxglove/ws-protocol-examples@latest mcap-play ./2023-08-08_10-45-04.461_measurement.mcap --loop
```

In Studio, open the Websocket connection. Create a 3D panel, and try to visualize `lidar_topic`.
Sit back, releax, and watch the RAM usage grow.

**Expected Behavior**
No memory leak.

foxymiles, 2023-08-17:
Does this happen when you leave the tab active or only when you switch away to another tab for a while then switch back?

foxymiles, 2023-08-18:
Oh sorry this is in the desktop app not a browser right?

KerstinKeller, 2023-08-19:
Yes, this is on the Desktop App, I didn't try on the browser.
Did you try and reproduce? It should be straightforward with the measurement I provided above.
And I think this even happened when the App window was active.

KerstinKeller, 2023-08-21:
https://github.com/foxglove/studio/assets/13848742/ccc25332-a601-4cf3-975c-ac97c649ddef

Maybe this video helps. It's a new layout, with only the 3D panel.
Note, that on the top, Foxglove doesn't report that the memory is increasing (but task manager does). Eventually it will lead to a crash when foxglove aquired ~16GB of memory (on my system with 32GB available).

foxymiles, 2023-08-21:
I can reproduce this on a mac now too. The memory shows up in Electron Helper in activity monitor, not within the browser window itself:

[image]

foxymiles, 2023-08-21:
I'm also seeing this happen with point cloud mcaps that don't have errors so I think the error reported in the 3d panel may actually be unrelated to the core problem.

foxymiles, 2023-08-21:
I can also reproduce this with nuscenes data and 4 3d panels.

achim-k, 2023-11-16:
Fixed by #7059

closed 2023-11-16 by achim-k