foxglove/studio #6617: 3D panel stopped working: THREE.WebGLProgram: Shader Error 0 - VALIDATE_STATUS false

jplapp, 2023-08-09:
**Description**
3d panel not working, error THREE.WebGLProgram: Shader Error 0 - VALIDATE_STATUS false

- Version: I tried 1.60.2 , 1.62, 1.63, 1.65
- Platform: ubuntu 22
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): all sources

**Steps To Reproduce**
Open any rosbag
Open 3d panel
-> no 3d shown
-> errors in logs 
[image]

**Expected Behavior**
3d shown

jtbandes, 2023-08-15:
> any rosbag I tried

This isn't really enough information for us to be able to reproduce this issue. Could you provide more details?

jtbandes, 2023-08-25:
Going to close out this ticket since we haven't gotten any response, but feel free to comment with more details and we can reopen it.

closed 2023-08-25 by jtbandes