foxglove/studio #6645: Received "Error initializing: Cannot identify bag format" error on valid bagfile.

jon-herman, 2023-08-15:
**Description**

App errors on opening local bagfile. 

[image]

I have confirmed the bag is valid by successfully opening in Plotjuggler and inspecting via ROS2 CLI. I have also confirmed my Foxglove installation by successfully loading other bagfiles.

Below is the output from the CLI:
```
$ ros2 bag info 05fe8ed3-e245-43b8-a242-214e91d57721_488875a8-9cbb-4110-9171-550cdc111d7f.bag -s mcap

Files:             05fe8ed3-e245-43b8-a242-214e91d57721_488875a8-9cbb-4110-9171-550cdc111d7f.bag
Bag size:          260.9 MiB
Storage id:        mcap
Duration:          2.827s
Start:             Aug 11 2023 19:39:19.749 (1691782759.749)
End:               Aug 11 2023 19:39:22.576 (1691782762.576)
Messages:          1382
Topic information: 
                   Topic: /rosout | Type: rcl_interfaces/msg/Log | Count: 35 | Serialization Format: cdr
                   Topic: /tf_static | Type: tf2_msgs/msg/TFMessage | Count: 46 | Serialization Format: cdr
                   ...
```

Thank you for your help!

- Version: 1.65.0
- Platform: Ubuntu 20.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): mcap file

**Steps To Reproduce**

Select file: `File -> Open local file...`

**Expected Behavior**
Load bagfile.

jtbandes, 2023-08-15:
Please share the bag file, otherwise there's not really much we can do here to diagnose the issue.

jon-herman, 2023-08-16:
Thank you for the prompt reply.

In the process of gathering my data to share, I found the problem. My bag had `mcap` storage but a `.bag` extension. After changing the extension the file opens.

I'll close this ticket.

closed 2023-08-16 by jon-herman