foxglove/studio #6654: v1.66 freezes after a few seconds - ubuntu 22.04

iftahnaf, 2023-08-16:
**Description**
the app freezes a few seconds after booting.

- Version: 1.66
- Platform: ubuntu 22.04
- Data source (e.g. bag file, mcap file, rosbridge, ROS 1/2 native): main app

**Steps To Reproduce**
```
$foxglove-studio
initializing Sentry in main
Foxglove Studio 1.66.0
Could not set app as handler for foxglove://
Elapsed (ms) until new StudioWindow: 58
Checking for beta autoupdate feature for deb/rpm distributions
Found package-type: deb
New Foxglove Studio window 1
Starting update loop
window.loadURL(file:///opt/Foxglove Studio/resources/app.asar/renderer/index.html)
[22780:0816/101836.753213:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 1 times!
load-pending-files
injectFilesToOpen: no files - skipping
[22780:0816/101837.177342:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 2 times!
window URL loaded
[22780:0816/101837.204880:ERROR:gl_surface_presentation_helper.cc(260)] GetVSyncParametersIfAvailable() failed for 3 times!

```

achim-k, 2023-08-16:
> the app freezes a few seconds after booting.

It freezes without you loading any mcap/bag file or connecting to a live data?

Is this a snap install or from apt?

iftahnaf, 2023-08-16:
> > the app freezes a few seconds after booting.
> 
> It freezes without you loading any mcap/bag file or connecting to a live data? 
> 
> Is this a snap install or from apt? apt. it worked fine until the update to 1.66

Yes, it freezes without loading or connecting to anything.
Installed with apt, worked fine until 1.66

achim-k, 2023-08-16:
I have just tested this locally but I can't reproduce the freezing behavior. Could you try again after temporarily renaming the config folder `~/.foxglove-studio` such that Studio doesn't find it?

iftahnaf, 2023-08-17:
The path is `~/.foxglove-studio`? I got nothing there.
I tried to reinstall with `apt` but it still not working.

YazanRihan, 2023-08-17:
I have the same issue and that was after the the recent update through apt

achim-k, 2023-08-17:
When the app freezes, can you still open the developer tools (<kbd>ctrl</kbd>+<kbd>shift</kbd>+<kbd>i</kbd>)? 
What are the memory / cpu stats of the process?

blauenase, 2023-08-17:
My frozen app uses 0,0% CPU and ~150MB Memory. I can open the Dev Tools window on startup, but after the crash it says: "DevTools was disconnected from the page". I can close but not open the DevTools after the crash.

achim-k, 2023-08-17:
We have upgraded the electron version to 26.0 in the last release, it might be that.

Could you check if an empty electron window already freezes on your system? With https://www.electronjs.org/fiddle one can check for various electron versions:

Link to gist used in the gif below: https://gist.github.com/achim-k/40e54478a617c46fd83e320f71e2ced6

[image]

blauenase, 2023-08-17:
The test with your posted gist and with electron 26.0.0 is working without crashes

achim-k, 2023-08-18:
> I can open the Dev Tools window on startup, but after the crash it says: "DevTools was disconnected from the page". I can close but not open the DevTools after the crash.

Do you see anything suspicious in the dev tools javascript console before the crash? Does clearing the `localStorage` help?

closed 2023-08-18 by foxymiles

jtbandes, 2023-08-21:
@iftahnaf @blauenase  We've rolled back the electron version but it would still help a lot if you can provide more information about the hardware you are using (and other questions @achim-k posted above).

iftahnaf, 2023-08-21:
@jtbandes v1.67 works, thanks!
When the app froze, I saw nothing informative with `ctrl` + `shift` + `i`, not much CPU usage.
I'm using ASUS TUF f15, with RTX3070, i7 12 gen, and 16 GB RAM.

iftahnaf, 2023-08-21:
> Could you try enabling some [electron debug logging](https://www.electronjs.org/docs/latest/tutorial/application-debugging#v8-crashes) to see if there is any more info that appears when the app crashes?
> 
> Does it only crash if you perform certain actions in the UI (like loading a data source or choosing a layout)?

I can revert to v1.66 tomorrow (GMT +2 here) and try again.

It crashed immediately regardless of what I'm doing.

jtbandes, 2023-08-21:
You can also try deleting or renaming the `~/.config/Foxglove Studio` folder to reset the app to the default state

iliabaranov, 2023-08-21:
Note: I tried moving the above folder, though it reset the layout entirely, crash still occurs